nfcpy/nfcpy #122: Got { IOError: [Errno 19] No such device} on RC-S320+ArduinoYun

wangbin-ouhin, 2019-02-17:
I got a { IOError: [Errno 19] No such device} error when execute the test program.
Does nfcpy support RC-S320.Is it possible to figure out the reason?

**Error Detail**
loading rcs956 driver for usb:054c:01bb         <-- Debug info which I added
Traceback (most recent call last):
  File "read_idm.py", line 10, in <module>
    clf = nfc.ContactlessFrontend('usb')
  File "/usr/lib/python2.7/site-packages/nfc/clf/__init__.py", line 76, in __init__
    raise IOError(errno.ENODEV, os.strerror(errno.ENODEV))
IOError: [Errno 19] No such device

**Test Program**
```
import nfc
import binascii

def connected(tag):
    idm = binascii.hexlify(tag.idm)
    print(idm)
    return idm
clf = nfc.ContactlessFrontend('usb')
clf.connect(rdwr={'on-connect': connected}) # now touch a tag
clf.close()
```

**Environment**
Platform:RC-S320 , Arduino Yun
Python:2.7.3
nfcpy:0.13.5

**lsusb result:**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 003: ID 054c:01bb Sony Corp. FeliCa S320 [PaSoRi]
Bus 001 Device 004: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader`

**python -m nfc --search-tty result**
This is the 0.13.5 version of nfcpy run in Python 2.7.3
on Linux-3.3.8-mips-with-libc0
I'm now searching your system for contactless devices
loading rcs956 driver for usb:054c:01bb
ERROR:nfc.clf.transport:no bulk endpoints for read and write
Sorry, but I couldn't find any contactless device

nehpetsde, 2019-03-02:
Sony RC-S320 is not supported.

wangbin-ouhin, 2019-03-03:
> Sony RC-S320 is not supported.

Thank you for the reply.I will try the RC-S380.

closed 2019-03-04 by nehpetsde

wangbin-ouhin, 2019-03-04:
Dear nehpetsde,
I tried RC-S380, but it didn't work.
Is there anything wrong with my environment.
Sorry for take your time.

**lsusb**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 005: ID 054c:06c3 Sony Corp. 
Bus 001 Device 006: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader

**python -m nfc --search-tty --verbose**
This is the 0.13.5 version of nfcpy run in Python 2.7.3
on Linux-3.3.8-mips-with-libc0
I'm now searching your system for contactless devices
INFO:nfc.clf:searching for reader on path usb:001:005
loading rcs380 driver for usb:054c:06c3
ERROR:nfc.clf.transport:USBErrorPipe()
ERROR:nfc.clf.transport:USBErrorPipe()
Sorry, but I couldn't find any contactless device

nehpetsde, 2019-03-10:
Try to see the debug output with '-v -v -v' and look for USB library errors.

wangbin-ouhin, 2019-03-12:
@nehpetsde Thank you for the reply.Here is the result.

**python -m nfc --search-tty --verbose -v -v -v**
This is the 0.13.5 version of nfcpy run in Python 2.7.3
on Linux-3.3.8-mips-with-libc0
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.9
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:001:003
DEBUG:nfc.clf.transport:using libusb-1.0.9
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
loading rcs380 driver for usb:054c:06c3
Level 9:nfc.clf.transport:>>> 0000ff00ff00
Level 9:nfc.clf.rcs380:SetCommandType 01
Level 9:nfc.clf.transport:>>> 0000ffffff0300fdd62a01ff00
ERROR:nfc.clf.transport:USBErrorPipe()
DEBUG:nfc.clf.device:[Errno 5] Input/output error
DEBUG:nfc.clf.transport:^(tty(S|ACM|AMA|USB)\d+|cu\.usbserial.*)$
DEBUG:nfc.clf.transport:check: /dev/ttyS0 /dev/ttyS1 /dev/ttyS2 /dev/ttyS3 /dev/ttyS4 /dev/ttyS5 /dev/ttyS6 /dev/ttyS7 /dev/ttyS8 /dev/ttyS9 /dev/ttyS10 /dev/ttyS11 /dev/ttyS12 /dev/ttyS13 /dev/ttyS14 /dev/ttyS15
DEBUG:nfc.clf.transport:avail: 
Sorry, but I couldn't find any contactless device

 **lsusb**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 003: ID 054c:06c3 Sony Corp. 
Bus 001 Device 004: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader

[transport:USBErrorPipe] I googled it ,but got nothing about this.
Is it possible to debug it further?

Best regards,

nehpetsde, 2019-03-12:
The libusb-1.0.9 version dates back to April 2012. Nfcpy started to use libusb about 4 years later with libusb-1.0.20. It seems very likely that this ancient version of libusb behaves differently but fundamentally it's impossible to tell. If possible, try upgrade libusb to some newer version.

wangbin-ouhin, 2019-03-12:
Thank you for the reply.I will try to update the libusb.