nfcpy/nfcpy #123: OSX - permission denied

thongly, 2019-02-28:
I am running into a similar program as [issue #115](https://github.com/nfcpy/nfcpy/issues/115).  However, I am on a MacBook Pro (13-inch, 2017, Four Thunderbolt 3 Ports).

Output of: `python -m nfc -vvv`

```
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:020:055
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading acr122 driver for usb:072f:2200
DEBUG:nfc.clf.device:[Errno 13] Permission denied
** found usb:072f:2200 at usb:020:055 but access is denied
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device
```

I am also using a 

Output of: `lsubs`

```
...
Bus 020 Device 055: ID 072f:2200 072f ACR122U PICC Interface
```

thongly, 2019-02-28:
Update... I am tracing this back to some OSX issues related to access of the USB.  Will update here for others in the future.

Output of: `nfc-list`

```nfc-list uses libnfc 1.7.1
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:020:006
nfc-list: ERROR: Unable to open NFC device: acr122_pcsc:ACS ACR122U```

thongly, 2019-02-28:
Ultimately, this seems to be related to the following:

- (Create a codeless kext file)[https://github.com/nfc-tools/libnfc/issues/330#issuecomment-169594231]

nehpetsde, 2019-03-01:
Access denied is usually either insufficient permissions or some other process that uses the device. Almost always a PC/SC daemon that you'll need to get out of the way.

thongly, 2019-03-03:
I didn't manage to get this working on OSX, despite installing and re-installing the drivers, but simply moved to an Ubuntu box that will be used in the end anyway.

However, others in the linked issue were able to solve this issue, so thought I close this and leave it here for others in the future.

closed 2019-03-03 by thongly

mpoirier-enac, 2019-03-15:
hi, 
I was developping something using 2 different macs
everything was working well since 2 days ago ... now i have the same probleme on each of my computer.

Despite a red led on the device (acr122)

```
$ pyhton main.py
with nfc.ContactlessFrontend('usb') as clf:
  File "/usr/local/lib/python2.7/site-packages/nfc/clf/__init__.py", line 75, in __init__
    raise IOError(errno.ENODEV, os.strerror(errno.ENODEV))
IOError: [Errno 19] Operation not supported by device
```
```

$python -m nfc -vvv
This is the 0.13.5 version of nfcpy run in Python 2.7.15
on Darwin-18.2.0-x86_64-i386-64bit
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:020:005
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading acr122 driver for usb:072f:2200
DEBUG:nfc.clf.device:[Errno 13] Permission denied
** found usb:072f:2200 at usb:020:005 but access is denied
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device
```

time to time it's working again ... but nothing i can forecast !

any suggestion ?

thongly, 2019-03-15:
> hi,
> I was developping something using 2 different macs
> everything was working well since 2 days ago ... now i have the same probleme on each of my computer.
>
> Despite a red led on the device [...]

To confirm, I never got this working on OSX, but a switch to a linux box was relatively straight forward.  You may also be able to install the NFC reader's drivers on OSX properly if you battle with it for a bit.

mpoirier-enac, 2019-03-15:
i found the solution here: 
https://github.com/nfc-tools/libnfc/issues/330#issuecomment-285668359

we have to disable com.apple.ifdreader ( -- SmartCard reader daemon)

mpoirier-enac, 2019-03-15:
it seems that I spoke too fast.

it seems to work with my Mojave macos but not on my high sierra macos

mpoirier-enac, 2019-03-18:
On my high sierra the service com.apple.ifdreader is automaticaly reload by another one call : CryptotokenKit. (in charge by security services )
https://developer.apple.com/documentation/cryptotokenkit
this is use to login using smartcard on macos ... which i really don't care 

but I can't manage to shut it down definitivly ... (I try several things)

now, my script, find the pid process of this service and kill it before launching my app.

it's working ... but...

nospam2000, 2019-12-08:
> we have to disable com.apple.ifdreader ( -- SmartCard reader daemon)

On stackoverflow I described a permanent solution by removing the USB device IDs from the SmartCardServices. See [this article](https://stackoverflow.com/a/59230125/4330024).