nfcpy/nfcpy #124: Can't get IDm from my Android mobile.

mb-arita, 2019-02-28:
I can't get IDm from my Android mobile phone on 0.13.5.

Result of executing tagtool.py without option

Type4ATag MIU=255 FWT=0.038664

Only the above was acquired.

https://github.com/nfcpy/nfcpy/issues/71

I performed the following with reference to the above.

tagtool.py --technology f

However, we could not get the data.

How can I get the IDm of Type 4 Tag?

nehpetsde, 2019-03-01:
IDm is a Type 3 Tag property and that is part of Felica technology. Fixing technology to 'F' is generally the right thing to do but it may still be that Felica is disabled on your phone. You could try with 'sense.py' to fire Felica probe requests and check if there's any response.

rennyaaru, 2019-03-02:
Dear nehpetsde.

Thank you for your answer.
Based on the keyword that Felica is invalid, I investigated the corresponding terminal.
**Felica was not installed in Sim free Xperia Xzs.**
For that reason, I believe that it is a correct result that IDm can not be acquired.
Prepare a Felica-equipped smartphone and try again.
Thank you very much.

closed 2019-03-02 by rennyaaru