nfcpy/nfcpy #125: Unreliability in clf.connect() for HCE tags

mclean25, 2019-04-13:
Hi there,

I'm using a PN532 breakout board with raspberry pi 3.

I seem to be having a lot of issues connecting to HCE tags using the following:
```
clf = nfc.ContactlessFrontend("ttyS0")
tag = clf.connect(rdwr=('on-connect': lambda tag: False)
```
It will just continue without finding any tag. If I use a spare type 2 tag I have laying around, it picks it up no problem:

```
>>> clf = nfc.ContactlessFrontend("ttyS0")
>>> tag = clf.connect(rdwr={'on-connect': lambda tag: False})
>>>
>>> print(tag)
Type2Tag ID=46DFB17C

```
 I thought the issue might be with the Android devices I have or the board, but if I use `libnfc` I have no issues:

```
pi@raspberrypi:~ $ ~/libnfc/libnfc-1.7.1/examples/nfc-poll
/home/pi/libnfc/libnfc-1.7.1/examples/.libs/lt-nfc-poll uses libnfc 1.7.1
NFC reader: pn532_uart:/dev/ttyS0 opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID3): 08  69  a3  2a
      SAK (SEL_RES): 60
                ATS: 78  80  70  02
nfc_initiator_target_is_present: Target Released
```

mclean25, 2019-04-14:
While I find the root of this issue quite suspicious, I was able to overcome it by using the following:

```
with nfc.ContactlessFrontend('tty:S0:pn532') as clf:
    tag = clf.connect(rdwr={'on-connect': lambda tag: False})
    print("connected to tag")
    print(tag)
```

along with updating the OS on the Android device (was a few months out of date; now on Android 9 with January 2019 security update).

Will close this issue for now, although I feel like there's something else going on here.

closed 2019-04-14 by mclean25

TE-StephenTiedemann, 2019-06-07:
In nfcpy you'll find a similar nfc-poll tool that you may use to inspect your phone's behavior. Answers generally depend on which target technologies the phone has enabled (A/B/F).

`examples/sense.py 106A 106B 212F 424F`

nehpetsde, 2019-06-07:
In nfcpy you'll find a similar nfc-poll tool that you may use to inspect your phone's behavior. Answers generally depend on which target technologies the phone has enabled (A/B/F).

`examples/sense.py 106A 106B 212F 424F`