nfcpy/nfcpy #129: Tagtool unable to format type 2 and 4 tags

michaelfaith84, 2019-05-16:
Missing feature.

nehpetsde, 2019-05-17:
Formatting NFC Forum compliant Type 2 and 4 tags is supported by nfcpy and it should be possible to use this with tagtool.py. If it is not working then either your tags are not NFC Forum compliant or you've hit a bug in nfcpy/tagtool. In the latter case please file an issue with sufficient context.

michaelfaith84, 2019-05-17:
If you look at the source of tagtool you'll see that the code is simply not there for formatting the aforementioned tag types. I used NXP's android app to format the tags after I noticed that. It worked perfectly.

nehpetsde, 2019-06-04:
Formatting an existing NDEF partition is done [in this line](https://github.com/nfcpy/nfcpy/blob/e3dd87850644d352bc51db9b93907d4ceac7c2f3/examples/tagtool.py#L390). The `format_tt1/2/3/4_tag` methods do extra stuff for messing with the NDEF management data, mostly for testing NDEF reader software.

michaelfaith84, 2019-06-04:
python tagtool.py --device tty:S0 format
No handlers could be found for logger "nfc.llcp.sec"
[nfc.clf] searching for reader on path tty:S0
[nfc.clf] using PN532v1.6 at /dev/ttyS0
** waiting for a tag **
[nfc.tag.tt4] no ndef capability file
[nfc.tag.tt4] format error: no ndef or not writeable
Sorry, I could not format this tag.

python tagtool.py --device tty:S0 -v
No handlers could be found for logger "nfc.llcp.sec"
[nfc.clf] searching for reader on path tty:S0
[nfc.clf] using PN532v1.6 at /dev/ttyS0
** waiting for a tag **
Type4ATag MIU=63 FWT=0.077329
[nfc.tag.tt4] no ndef capability file
Memory Dump:
[nfc.tag.tt4] no ndef capability file

Look closer at the source:

def format_tt2_tag(self, tag):
    pass

def format_tt4_tag(self, tag):
    pass

closed 2019-07-07 by michaelfaith84