nfcpy/nfcpy #131: Confusion between Type2Tag Mifare Ultralight (MF01CU1) and Type2Tag NTAG203

LeSanchez, 2019-06-05:
I'm trying to read/write data on the [NTAG203](https://www.adafruit.com/product/2800l) with a [PN532 reader](https://www.adafruit.com/product/364). I run the following commands
```
>>> tag = clf.connect(rdwr={'on-connect': lambda tag: False})
>>> print tag
Type2Tag 'Mifare Ultralight (MF01CU1)' ID=04087A04042019
```

After some research I realized this wasn't the tag I had bought. Read/Write operations still work but required a `tag.format()` too function.
 Is there a way I can force it to realize it's an NTAG203 and not a MF01CU1?

Thank you

nehpetsde, 2019-06-05:
Please run `tagtool.py -d nfc.tag` and look for the messages printed from [the tag activation function](https://github.com/nfcpy/nfcpy/blob/a8c7b1bf658ef55757f1d4fe3ee43ec1a2e8db2d/src/nfc/tag/tt2_nxp.py#L732). step through the code and check why [this line](https://github.com/nfcpy/nfcpy/blob/a8c7b1bf658ef55757f1d4fe3ee43ec1a2e8db2d/src/nfc/tag/tt2_nxp.py#L756) is not executed.

LeSanchez, 2019-06-06:
I just step through the code and it seems like the [line 749]
(https://github.com/nfcpy/nfcpy/blob/a8c7b1bf658ef55757f1d4fe3ee43ec1a2e8db2d/src/nfc/tag/tt2_nxp.py#L749) `rsp = bytes(clf.exchange(b'\x60', timeout=0.01))`
did not return. I get a timeout exception:

[image]

LeSanchez, 2019-06-06:
The more I think about it, I realize I could probably just hardcode the tag type in my own code since this tag is literally the only one that will ever need to connect to my reader. I just don't know if this could harm my tag  in any way. Do you think this would work?  @nehpetsde

nehpetsde, 2019-06-07:
You can certainly just hardcode the choice but still I'd be interested to know why its not properly detected. If possible, could you run `examples/tagtool.py -d nfc.clf -v --reltime` and post the debug log (place the tag on the reader before running tagtool, it's producing lots of output while waiting for a tag).

LeSanchez, 2019-06-07:
Thank you for your time, here's the log you asked for. There really is lots of output :sweat_smile:.

```
python tagtool.py -d nfc.clf --device tty:AMA0 -v --reltime 
No handlers could be found for logger "nfc.llcp.sec"
508 ms [main] enable debug output for 'nfc.clf'
509 ms [nfc.clf] searching for reader on path tty:AMA0
511 ms [nfc.clf.transport] ^ttyAMA0$
512 ms [nfc.clf.transport] check: /dev/ttyAMA0
513 ms [nfc.clf.transport] avail: /dev/ttyAMA0
513 ms [nfc.clf.device] trying arygon on /dev/ttyAMA0
623 ms [nfc.clf.device] [Errno 19] No such device
624 ms [nfc.clf.device] trying pn532 on /dev/ttyAMA0
625 ms [nfc.clf.pn532] running on Raspberry Pi 3 Model B Rev 1.2
626 ms [nfc.clf.transport] >>> 000000000000000000000000ff02fed4022a00
626 ms [nfc.clf.pn532] wait 100 ms for data on /dev/ttyAMA0
630 ms [nfc.clf.transport] <<< 0000ff00ff00
631 ms [nfc.clf.transport] <<< 0000ff06fad50332010607e800
631 ms [nfc.clf.transport] >>> 000000000000000000000000ff05fbd4140100001700
635 ms [nfc.clf.transport] <<< 0000ff00ff00
636 ms [nfc.clf.transport] <<< 0000ff02fed5151600
637 ms [nfc.clf.pn532] trying to set 921600 baud
651 ms [nfc.clf.transport] >>> 000000000000000000000000ff03fdd410071500
655 ms [nfc.clf.transport] <<< 0000ff00ff00
657 ms [nfc.clf.transport] <<< 0000ff02fed5111a00
658 ms [nfc.clf.transport] >>> 0000ff00ff00
686 ms [nfc.clf.pn532] changed uart speed to 921600 baud
689 ms [nfc.clf.pn532] Diagnose 00000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fafbfcfdfeff000102030405 1.000s
691 ms [nfc.clf.transport] >>> 0000ffffff0109f6d40000000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fafbfcfdfeff0001020304059d00
697 ms [nfc.clf.transport] <<< 0000ff00ff00
739 ms [nfc.clf.transport] <<< 0000ffffff0109f6d50100000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fafbfcfdfeff0001020304059b00
740 ms [nfc.clf.pn532] GetFirmwareVersion  0.100s
741 ms [nfc.clf.transport] >>> 0000ff02fed4022a00
742 ms [nfc.clf.transport] <<< 0000ff00ff00
742 ms [nfc.clf.transport] <<< 0000ff06fad50332010607e800
743 ms [nfc.clf.pn532] chipset is a PN532v1.6
744 ms [nfc.clf.pn532] SetParameters 00 0.100s
744 ms [nfc.clf.transport] >>> 0000ff03fdd412001a00
745 ms [nfc.clf.transport] <<< 0000ff00ff00
746 ms [nfc.clf.transport] <<< 0000ff02fed5131800
747 ms [nfc.clf.pn532] RFConfiguration 02000b0a 0.100s
747 ms [nfc.clf.transport] >>> 0000ff06fad43202000b0ae300
749 ms [nfc.clf.transport] <<< 0000ff00ff00
750 ms [nfc.clf.transport] <<< 0000ff02fed533f800
750 ms [nfc.clf.pn532] RFConfiguration 0400 0.100s
751 ms [nfc.clf.transport] >>> 0000ff04fcd4320400f600
752 ms [nfc.clf.transport] <<< 0000ff00ff00
753 ms [nfc.clf.transport] <<< 0000ff02fed533f800
754 ms [nfc.clf.pn532] RFConfiguration 05010001 0.100s
755 ms [nfc.clf.transport] >>> 0000ff06fad43205010001f300
758 ms [nfc.clf.transport] <<< 0000ff00ff00
759 ms [nfc.clf.transport] <<< 0000ff02fed533f800
760 ms [nfc.clf.pn532] write analog settings for Type A 106 kbps
760 ms [nfc.clf.pn532] RFConfiguration 0a59f43f114d85616f266287 0.100s
761 ms [nfc.clf.transport] >>> 0000ff0ef2d4320a59f43f114d85616f266287a200
762 ms [nfc.clf.transport] <<< 0000ff00ff00
765 ms [nfc.clf.transport] <<< 0000ff02fed533f800
766 ms [nfc.clf.pn532] write analog settings for Type F 212/424 kbps
767 ms [nfc.clf.pn532] RFConfiguration 0b69ff3f114185616f 0.100s
768 ms [nfc.clf.transport] >>> 0000ff0bf5d4320b69ff3f114185616fa100
769 ms [nfc.clf.transport] <<< 0000ff00ff00
771 ms [nfc.clf.transport] <<< 0000ff02fed533f800
772 ms [nfc.clf.pn532] write analog settings for Type B 106 kbps
772 ms [nfc.clf.pn532] RFConfiguration 0cff0485 0.100s
773 ms [nfc.clf.transport] >>> 0000ff06fad4320cff04856600
774 ms [nfc.clf.transport] <<< 0000ff00ff00
775 ms [nfc.clf.transport] <<< 0000ff02fed533f800
776 ms [nfc.clf.pn532] write analog settings for 14443-4 212/424/848 kbps
776 ms [nfc.clf.pn532] RFConfiguration 0d85158a8508b28501da 0.100s
777 ms [nfc.clf.transport] >>> 0000ff0cf4d4320d85158a8508b28501da2a00
[... 556 lines cut ...]
  003: e1 10 12 00 (OTP0-OTP3)
  004: 01 03 a0 10 |....|
  005: 44 03 40 91 |D.@.|
  006: 01 08 54 02 |..T.|
  007: 65 6e 73 61 |ensa|
  008: 6c 75 74 11 |lut.|
[... cut ...]
  00F: 43 6f 72 6f |Coro|
1475 ms [nfc.clf.transport] >>> 0000ff00ff00
1487 ms [nfc.clf.pn532] SetSerialBaudrate 04 0.100s
1488 ms [nfc.clf.transport] >>> 0000ff03fdd410041800
1489 ms [nfc.clf.transport] <<< 0000ff00ff00
1491 ms [nfc.clf.transport] <<< 0000ff02fed5111a00
1492 ms [nfc.clf.transport] >>> 0000ff00ff00
1494 ms [nfc.clf.pn532] PowerDown b000 0.100s
1495 ms [nfc.clf.transport] >>> 0000ff04fcd416b0006600
1498 ms [nfc.clf.transport] <<< 0000ff00ff00
1499 ms [nfc.clf.transport] <<< 0000ff03fdd517001400

```

nehpetsde, 2019-06-13:
Clearly this is not a Mifare Ultralight because it holds more data than would fit on that tag. But I've tested the detection logic again with different readers (including exactly an Adafruit PN532 board) and 8 different NTAG203 tags and it all worked perfect. Maybe you could try an Android phone to read your tag with the NXP TagInfo app and check it really is what you believe it to be.

LeSanchez, 2019-06-14:
First of all Thank you for your time this is really appreciated! 

I tried to read my tags with my phone but I  I can't read my them at all .... There seems to be something weird with them. perhaps it's there size? [These](https://www.adafruit.com/product/2800) are what I bough and they are fairly small so I could imagine how an antenna like that could fail to get picked up by a reader on a phone since there must be a lot of electromagnetic noise nearby. did you try with such small NTAG203? 

Also, I have another tag (NXP MIFARE Classik 1k) which is in the form of a normal sized card. And this one works like a charm I can read and write from an android phone without any issues.

Thanks again

nehpetsde, 2019-06-17:
I haven't tested those tiny tags as sold by Adafruit. Issues with phone reading may well be to the very small antenna size but I doubt that to be a problem with your PN532 read as shown by the log. All commands complete with ~1ms except the VERSION command that times out after ~13 ms (this is a requested timeout). According to the NTAG203 specification it is supposed to answer immediately with an error code. And that is what my NTAGs do.
Unfortunately there is no easy way to get around here. Unless someone from NXP helps with a better explanation of a tag product identification procedure, the only solution might be that you just force the tag into the NATG203 type.

LeSanchez, 2019-06-18:
I was finally able to get a read on my tag by placing it in a very specific angle under my phone. It also says NXP Mifare Ultralight which means the issue is probably in the tag itself and not with this library. Thank you for your help!

closed 2019-06-18 by LeSanchez