nfcpy/nfcpy #132: tagtool not showing memory

BWhitten, 2019-06-14:
Hi all,

I have cloned the latest in github version 1.0.1 and setup on my Linux Mint virtual machine.
I am using PN532 over uart connected with an FTDI cable and I have confirmed that I can talk to the scanner.

When I run 'sudo ./tagtool.py -d nfc.tag -v --device tty:USB0:pn532 show'
I see that the tag is being read with the debug data and I have verified that the trancieve log output matches what I expect in the tag, however the memory is not displayed on the screen.

Bellow is the debug output for the tag

[main] enable debug output for 'nfc.tag'
[nfc.clf] searching for reader on path tty:USB0:pn532
[nfc.clf] using PN532v1.6 at /dev/ttyUSB0
** waiting for a tag **
[nfc.tag] trying to activate 106A sdd_res=028600071CA719 sel_res=20 sens_res=4200
[nfc.tag] trying type 4 tag activation for 106A
[nfc.tag.tt4] send RATS command to activate the Type 4A Tag
[nfc.tag.tt4] rcvd RATS response: 0578005002
[nfc.tag.tt4] max command frame size is 256 byte
[nfc.tag.tt4] max frame waiting time is 0.009666
Type4ATag MIU=255 FWT=0.009666
[nfc.tag.tt4] read ndef data
[nfc.tag.tt4] select ndef application
[nfc.tag.tt4] >> 00a4040007d2760000850101
[nfc.tag.tt4] << 9000
[nfc.tag.tt4] selected d2760000850101
[nfc.tag.tt4] select ndef capability file
[nfc.tag.tt4] >> 00a4000c02e103
[nfc.tag.tt4] << 9000
[nfc.tag.tt4] selected e103
[nfc.tag.tt4] read ndef capability file
[nfc.tag.tt4] read_binary from 0 to 2
[nfc.tag.tt4] >> 00b0000002
[nfc.tag.tt4] << 000f9000
[nfc.tag.tt4] read_binary from 2 to 15
[nfc.tag.tt4] >> 00b000020d
[nfc.tag.tt4] << 2000f600f604060001020000009000
[nfc.tag.tt4] ndef mapping version 2.0
[nfc.tag.tt4] max apdu response length 246
[nfc.tag.tt4] max apdu command length 246
[nfc.tag.tt4] ndef file control tlv tag 4
[nfc.tag.tt4] ndef file identifier 0001
[nfc.tag.tt4] ndef file size limit 512
[nfc.tag.tt4] ndef file read flag is 0
[nfc.tag.tt4] ndef file write flag is 0
[nfc.tag.tt4] select ndef data file
[nfc.tag.tt4] >> 00a4000c020001
[nfc.tag.tt4] << 9000
[nfc.tag.tt4] selected 0001
[nfc.tag.tt4] read ndef data file
[nfc.tag.tt4] read_binary from 0 to 2
[nfc.tag.tt4] >> 00b0000002
[nfc.tag.tt4] << 9f009000
[nfc.tag.tt4] ndef data length is 40704
[nfc.tag.tt4] read_binary from 2 to 248
[nfc.tag.tt4] >> 00b00002f6
[nfc.tag.tt4] << 0fc014c180c200c380007fffc501c600c70ac800c900ca5c6ee605cb00cf8000000000000002a7c25e23f15e4dff98c33ceaf3013dc9d004d100d2040404d301d400d500d600d700d90cda18db00dc00dd000000de0cdf0be008e1cc5314557119544d4ed80c75f888c8f0e29e6e2373723c12f60c4897ee600e4671e300000001e400000000e5cde61fe700000000e800000000ee01efa01491fffe0001380000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000009000
[nfc.tag.tt4] read_binary from 248 to 494
[nfc.tag.tt4] >> 00b000f8f6
[nfc.tag.tt4] << 0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000009000
[nfc.tag.tt4] read_binary from 494 to 740
[nfc.tag.tt4] >> 00b001eef6
[nfc.tag.tt4] << 6282
Memory Dump:
[nfc.tag.tt4] read ndef data
[nfc.tag.tt4] select ndef application
[nfc.tag.tt4] >> 00a4040007d2760000850101
[nfc.tag.tt4] << 9000
[nfc.tag.tt4] selected d2760000850101
[nfc.tag.tt4] select ndef capability file
[nfc.tag.tt4] >> 00a4000c02e103
[nfc.tag.tt4] << 9000
[nfc.tag.tt4] selected e103
[nfc.tag.tt4] read ndef capability file
[nfc.tag.tt4] read_binary from 0 to 2
[nfc.tag.tt4] >> 00b0000002
[nfc.tag.tt4] << 000f9000
[nfc.tag.tt4] read_binary from 2 to 15
[nfc.tag.tt4] >> 00b000020d
[nfc.tag.tt4] << 2000f600f604060001020000009000
[nfc.tag.tt4] ndef mapping version 2.0
[nfc.tag.tt4] max apdu response length 246
[nfc.tag.tt4] max apdu command length 246
[nfc.tag.tt4] ndef file control tlv tag 4
[nfc.tag.tt4] ndef file identifier 0001
[nfc.tag.tt4] ndef file size limit 512
[nfc.tag.tt4] ndef file read flag is 0
[nfc.tag.tt4] ndef file write flag is 0
[nfc.tag.tt4] select ndef data file
[nfc.tag.tt4] >> 00a4000c020001
[nfc.tag.tt4] << 9000
[nfc.tag.tt4] selected 0001
[nfc.tag.tt4] read ndef data file
[nfc.tag.tt4] read_binary from 0 to 2
[nfc.tag.tt4] >> 00b0000002
[nfc.tag.tt4] << 9f009000
[nfc.tag.tt4] ndef data length is 40704
[nfc.tag.tt4] read_binary from 2 to 248
[nfc.tag.tt4] >> 00b00002f6
[nfc.tag.tt4] << 0fc014c180c200c380007fffc501c600c70ac800c900ca5c6ee605cb00cf8000000000000002a7c25e23f15e4dff98c33ceaf3013dc9d004d100d2040404d301d400d500d600d700d90cda18db00dc00dd000000de0cdf0be008e1cc5314557119544d4ed80c75f888c8f0e29e6e2373723c12f60c4897ee600e4671e300000001e400000000e5cde61fe700000000e800000000ee01efa01491fffe0001380000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000009000
[nfc.tag.tt4] read_binary from 248 to 494
[nfc.tag.tt4] >> 00b000f8f6
[nfc.tag.tt4] << 0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000009000
[nfc.tag.tt4] read_binary from 494 to 740
[nfc.tag.tt4] >> 00b001eef6
[nfc.tag.tt4] << 6282

BWhitten, 2019-06-14:
I also see that the ndef field isn't populated in the tag, I assume it is may be because its malformed in some way.
I expect I will need to read and write directly to the tag, but I'm not sure how to go about doing that with nfcpy.

nehpetsde, 2019-06-17:
The NDEF looks corrupt and the length is definitely wron. Try `tagtool.py format`.

closed 2019-07-27 by nehpetsde