nfcpy/nfcpy #133: support mifare classic 1k

johwick12, 2019-07-01:
Hi

I have a Sony RC-S380 NFC reader and would like to read Mifare Classic 1K cards. I know its not supported by nfcpy and I would like to know how I can add support for it using nfcpy.

nehpetsde, 2019-07-04:
Mifare Classic requires support for NXP's non-disclosed CRYPTO1 encryption, which RC-S380 doesn't have. I know that CRYPTO1 has been successfully reverse engineered but can't give any good pointers.

closed 2019-07-27 by nehpetsde

Shulyaka, 2020-11-29:
This tool reads MIFARE Classic cards just fine: https://github.com/nfc-tools/libnfc/blob/master/utils/nfc-mfclassic.c
Could it be ported to nfcpy?

nehpetsde, 2020-12-04:
I'll happily accept well written patches to add crypto1 support which should be possible to do for NXP reader ICs. I personally see those Mifare Classic tags as not standards compliant and won't spend time on them.