nfcpy/nfcpy #136: examples/tagtool.py emulation not reading the ndef file correctly

tts320, 2019-07-23:
I'm having difficulties getting the examples/tagtool.py working in the emulate mode. Writing a tag works fine. This is on Raspberry Pi Zero with Raspbian Jessie and Python 3.4.2. I'm using the master branch versions of both nfcpy and ndeftool. PN532 v1.6 NFC board. 

I think the issue might be related to the way Python3 handles argument character encoding. This page describes the issue: https://click.palletsprojects.com/en/7.x/python3/

I have the following locale settings:
LANG="en_US.UTF-8"
LC_ALL="en.US.UTF-8"
LC_CTYPE="UTF-8"

`
pi@raspberrypi:~ $ ndeftool text "Hello World" id "r1" uri "http://nfcpy.org" save -k "message.ndef" print

NDEF Text Record ID 'r1' Text 'Hello World' Language 'en' Encoding 'UTF-8'
NDEF Uri Record ID '' Resource 'http://nfcpy.org'

pi@raspberrypi:~ $ python3 nfcpy-master/examples/tagtool.py -v --device tty:AMA0:pn532 emulate message.ndef tt3
[nfc.clf] searching for reader on path tty:AMA0:pn532
[nfc.clf] using PN532v1.6 at /dev/ttyAMA0
Traceback (most recent call last):
  File "nfcpy-master/examples/tagtool.py", line 521, in prepare_tt3_tag
    self.options.data
AttributeError: 'Namespace' object has no attribute 'data'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "nfcpy-master/examples/tagtool.py", line 613, in <module>
    TagTool().run()
  File "/home/pi/nfcpy-master/examples/cli.py", line 375, in run
    while self.run_once() and self.options.loop:
  File "/home/pi/nfcpy-master/examples/cli.py", line 370, in run_once
    return clf.connect(**kwargs)
  File "/usr/local/lib/python3.4/dist-packages/nfcpy-1.0.1-py3.4.egg/nfc/clf/__init__.py", line 562, in connect
    target = card_options['on-startup'](target)
  File "nfcpy-master/examples/tagtool.py", line 323, in on_card_startup
    target = self.prepare_tag(target)
  File "nfcpy-master/examples/tagtool.py", line 512, in prepare_tag
    return self.prepare_tt3_tag(target)
  File "nfcpy-master/examples/tagtool.py", line 524, in prepare_tt3_tag
    self.options.data = self.options.input.read()
  File "/usr/lib/python3.4/codecs.py", line 313, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x99 in position 0: invalid start byte
`

mclean25, 2019-07-24:
Bump, I'm getting the same issue.

nehpetsde, 2019-07-24:
@mclean25 Also Python 3.4 or with a newer version?

mclean25, 2019-07-24:
@nehpetsde yes, Python 3.4 and the `master` of both `nfcpy` & `ndeftool` as of a couple of hours ago.

nehpetsde, 2019-07-24:
I can confirm that this is also failing on Python 3.5 and later.

nehpetsde, 2019-07-24:
Latest master should work. New release 1.0.2 follows when checks have passed.

nehpetsde, 2019-07-24:
Release 1.0.2 fixed this issue.

closed 2019-07-24 by nehpetsde