nfcpy/nfcpy #137: Emulating NDEF card error: AttributeError: 'Namespace' object has no attribute 'data'

mclean25, 2019-07-26:
I've got both the current `master` of both `nfcpy` and `ndeftool` on my machine. I'm trying to emulate a simple card with text on an rpi 3 with a pn532 breakout board. I'm creating my `.ndef` like so:

```
$ ndeftool text "Hello World" save --force ~/hello-world.ndef
$ python ./tagtool.py -l --device tty:S0 emulate ~/hello-world.ndef tt3
```

When I try to emulate it in Python 3.4 & 3.6, the following happens:
```
$ python3.6 ./tagtool.py -l --device tty:S0 emulate ~/hello-world.ndef tt3
[nfc.clf] searching for reader on path tty:S0
[nfc.clf] using PN532v1.6 at /dev/ttyS0
Traceback (most recent call last):
  File "./tagtool.py", line 521, in prepare_tt3_tag
    self.options.data
AttributeError: 'Namespace' object has no attribute 'data'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tagtool.py", line 613, in <module>
    TagTool().run()
  File "/home/pi/git/nfcpy/examples/cli.py", line 375, in run
    while self.run_once() and self.options.loop:
  File "/home/pi/git/nfcpy/examples/cli.py", line 370, in run_once
    return clf.connect(**kwargs)
  File "/usr/local/lib/python3.6/site-packages/nfc/clf/__init__.py", line 562, in connect
    target = card_options['on-startup'](target)
  File "./tagtool.py", line 323, in on_card_startup
    target = self.prepare_tag(target)
  File "./tagtool.py", line 512, in prepare_tag
    return self.prepare_tt3_tag(target)
  File "./tagtool.py", line 524, in prepare_tt3_tag
    self.options.data = self.options.input.buffer.read()
AttributeError: '_io.BufferedReader' object has no attribute 'buffer'
```

Python 2 yields a similar result:
```
$ python ./tagtool.py  --device tty:S0 emulate ~/hello-world.ndef tt3
[nfc.clf] searching for reader on path tty:S0
[nfc.clf] using PN532v1.6 at /dev/ttyS0
Traceback (most recent call last):
  File "./tagtool.py", line 613, in <module>
    TagTool().run()
  File "/home/pi/git/nfcpy/examples/cli.py", line 375, in run
    while self.run_once() and self.options.loop:
  File "/home/pi/git/nfcpy/examples/cli.py", line 370, in run_once
    return clf.connect(**kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/nfc/clf/__init__.py", line 562, in connect
    target = card_options['on-startup'](target)
  File "./tagtool.py", line 323, in on_card_startup
    target = self.prepare_tag(target)
  File "./tagtool.py", line 512, in prepare_tag
    return self.prepare_tt3_tag(target)
  File "./tagtool.py", line 524, in prepare_tt3_tag
    self.options.data = self.options.input.buffer.read()
AttributeError: 'file' object has no attribute 'buffer'
```

mclean25, 2019-07-26:
This looks like it was caused by this commit? https://github.com/nfcpy/nfcpy/commit/2caf9cdf45f5fcabafd0f9b1658daf8f042466b5

nehpetsde, 2019-07-26:
@mclean25 That is unfortunately correct, thanks for testing.

nehpetsde, 2019-07-27:
@mclean25 It should be working now with latest master (v1.0.3), please try. Note that tagtool.py now requires Python3 (by version check and sys.exit if running with Python2).

mclean25, 2019-08-25:
Sorry for the late reply, this is resolved now. Works great.

closed 2019-08-25 by mclean25