nfcpy/nfcpy #141: RCS380 reader Field On with unexpected modulated signal(start of communication)

ghxbob, 2019-08-26:
Hi,

When i send readerFieldOn with RCS380 reader, i always got an unexpected modulated signal(start of communication), could anyone explain why this happens.
Is it a bug of the firmware of RCS380.

PS. i use an external RF spy to get the physical layer of the traces, the tool is reliable and trusty.

The class i'm using is rcs380, and the function i'm calling is

    def switch_rf(self, switch):
        switch = ("off", "on").index(switch)
        data = self.send_command(0x06, [switch], 100)
        if data and data[0] != 0:
            raise StatusError(data[0])

ghxbob, 2019-08-27:
any comment?

nehpetsde, 2019-08-29:
Which of the two functions are you then calling? For in_comm_rf() it would be logical to see start of communication after some initial guard time. And that should be followed by the data argument bytes.

For switch_rf("on") I'd not really expect anything else than a carrier activation, but in nfcpy only uses switch_rf("off") and I thus can't tell.

mofe23, 2019-09-01:
@nehpetsde had no time to confirm this issue but we rely heavily on those Sony readers for our internal development. The reason for us to chose those was the availability and your employment at Sony which made me confident after spending some time with your code (as you know :)) that those will work for us, even if not a single page of documentation is available.

Are you in a position to lobby a bit at your employer to release some docs?