nfcpy/nfcpy #142: Python3 Pycharm WinError126

GensokyoLRE, 2019-08-31:
So, I've been trying to use nfcpy now that nfcpy works with 3.0, but... I'm getting this:
`Traceback (most recent call last):
  File "D:/_[omitted]_/Project2/Program2.py", line 9, in <module>
    import nfc
  File "D:\_[omitted]_\venv\lib\site-packages\nfc\__init__.py", line 22, in <module>
    from . import clf                                                  # noqa: F401
  File "D:\_[omitted]_\venv\lib\site-packages\nfc\clf\__init__.py", line 25, in <module>
    from . import device
  File "D:\_[omitted]_\venv\lib\site-packages\nfc\clf\device.py", line 32, in <module>
    from . import transport
  File "D:\_[omitted]_\venv\lib\site-packages\nfc\clf\transport.py", line 31, in <module>
    import usb1 as libusb
  File "D:\_[omitted]_\venv\lib\site-packages\usb1\__init__.py", line 61, in <module>
    from . import libusb1
  File "D:\_[omitted]_\venv\lib\site-packages\usb1\libusb1.py", line 199, in <module>
    libusb = _loadLibrary()
  File "D:\_[omitted]_\venv\lib\site-packages\usb1\libusb1.py", line 173, in _loadLibrary
    return dll_loader('libusb-1.0' + suffix, **loader_kw)
  File "C:\_[omitted]_\Python\Python37-32\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found

Process finished with exit code 1
`

Am I missing something? Pycharm loaded all the prereq/necessary stuff along with nfcpy on module install, afaik.

The version of Pycharm:
PyCharm 2019.2.1 (Professional Edition)
Build #PY-192.6262.63, built on August 21, 2019

nehpetsde, 2019-08-31:
Libusb is missing. Did you follow the Installation instructions?

GensokyoLRE, 2019-09-02:
Well, now I have; I would've thought that PyCharm noting libusb being installed too meant it was installed to where it needed to be. I wasn't aware I had to still manually install it.

closed 2019-09-02 by GensokyoLRE