nfcpy/nfcpy #144: Reading NDEF data

ab3dat, 2019-10-07:
Hi,

I'm not sure whether this post should be posted here or better in the ndeflib repo, so I decided to post it here for now.

I have PN532 connected via UART to a Raspberry Pi 3B. In a Python script (V3.7.4) I activate the reader and read in the data of the tag. If there is a NDEF message, I read the NDEf data from it and return the records in a dictionary. My problem is, that checking the tag for a NDEF message doesn't work properly. When emulating a tag with the Android App NFC Tools Pro, the NDEF message could be read. Using various other apps, the NDEF message could not be read. When I program a tag with NFC tools Pro, the NDEF message can also not be found. Using the other apps and the tags with a USB NFC reader or scanning a tag with the Phone, the NDEF data can be found and read. 

This is the code I use:

`
from time import sleep

import nfc
import ndef
from nfc.clf import RemoteTarget

def parseNDEF():
     with nfc.ContactlessFrontend("tty:S0:pn532") as clf:
        while True:
            target = clf.sense(RemoteTarget("106A"), RemoteTarget("106B"), RemoteTarget("212F"))
            print("Traget: "+str(target))
            
            if target is None:
                print("No target found ...")
                sleep(0.5)
                continue

            serial = target.sdd_res.hex()
            print("Found a target with serial " + serial + "!")

            tag = nfc.tag.activate(clf, target)
            print("tag data:\n" + str(tag.dump))

            nfcrecs = {}
            if tag.ndef:
                print("Tag is NDEF formatted!")
                print("It has " + str(len(tag.ndef.records)) + " records.")

                i=1;
                for record in tag.ndef.records:
                    print("Record: "+str(record))
                    nfcrecs["rec"+str(i)] = record.text
                    i+=1;

            else:
                print("No NDEF formatted tag.")

            if (len(nfcrecs)>0):
                return nfcrecs
'

ab3dat, 2019-10-10:
Here are some screenshots of the results I get.

Result of reading the data from the Android App NFC tools Pro with the PN532 on the Pi:
[image]

I programmed a chip with the same data I've sent with the app. Programming was also done with NFC tools Pro. This is the result of reading the chip with NFC tools Pro:
[image]

This is the result when I read the above programmed chip with the PN532 on the Pi:
[image]

Any idea what's going wrong here?

Thanks and regards.

kungfoomanchu, 2020-01-09:
I have the same problem. I only have this problem with MIFARE Classic cards and not NTAG cards, so it might be something to do with the MIFARE Classic format.

I would love to figure out how to read and write NDEF records to and from MIFARE Classic cards with nfcpy

Shulyaka, 2020-11-29:
This tool reads MIFARE Classic cards just fine: https://github.com/nfc-tools/libnfc/blob/216145fba522a8eaefc5c751b8c10198376bc066/utils/nfc-mfclassic.c
Could it be ported to nfcpy?