nfcpy/nfcpy #145: Serial is always changing

Bouni, 2019-10-30:
I try to identify a card by its serial ut the result is changing on every read.
My reader is a ACR-122U and I tried several contactless debit and credit card as well as my ID card.

```
import nfc
from nfc.clf import RemoteTarget

clf = nfc.ContactlessFrontend("usb")
target = clf.sense(RemoteTarget('106A'), RemoteTarget('106B'), RemoteTarget('212F'))
print(target)
if target:
    serial = target.sdd_res.hex()
    print(serial)
clf.close()
```

The result of my debit card for the first run:
```
106A sdd_res=0824A937 sel_res=20 sens_res=0400
0824a937
```

and for the second run:
```
106A sdd_res=080D3523 sel_res=20 sens_res=0400
080d3523
```

The result for my ID card lokks the same.

For my credit card on the other hand I get a constant serial:
```
106A sdd_res=A727411505A015 sel_res=20 sens_res=4400
a727411505a015
```

Can you point out what I do wrong here?
I tried `pyscard` before, with that I get consistent results but the usage sucks compared with nfcpy :-)

readysteadywhoa, 2019-11-30:
Might not be anything with your hardware or software. A lot of the more secure cards (Mifare DESfire, etc) have a random UID feature which generates a new UID every time it's powered up. It's used in a lot of bank cards to help prevent tracking the card itself through use.

If your project requires a UID that doesn't change, you'll need to use standard cards that don't have this feature.

Bouni, 2019-12-03:
Yes I tried several cards, some change their UID with the other lib as well (bank cards, ID card) but some seem to have a static UID. I figured out that A certain type of card gives me a static UID and I'm going to use them. 

Is there a way to dump the entire data on the card?
Maybe there is something on it that I can use to identify the card 🤔

nehpetsde, 2019-12-10:
The `examples/tagtool.py` script has a memory dump option but beware that credit cards are often encrypted.

Bouni, 2019-12-11:
Thanks for the hint, I'll take a look at it. Maybe that will do the trick!

nehpetsde, 2020-02-12:
I assume this is solved.

closed 2020-02-12 by nehpetsde