nfcpy/nfcpy #147: is elechouse PN532 V3 module supported?

fijar-lazuardy, 2019-12-07:
hi, i'm wanted to use your library for my project, and it doesn't work on my nfc module, i'ts exactly same like this https://github.com/nfcpy/nfcpy/issues/104 issue and try it (i go back to your commit 3 years ago) but still doesn't work, do you have any idea why?

closed 2019-12-10 by fijar-lazuardy

readysteadywhoa, 2019-12-10:
I've been using the Elechouse V3 (and now V4) modules in HSU mode with nfcpy for years. The biggest issue I've encountered is the module doesn't always like to close properly when the script exits, and needs to be hard-reset in order to communicate again. This appears to have been fixed with the V4 modules.

Aahil52, 2022-04-02:
Hey, it's been a few years, but I'm encountering the same issue. I haven't been able to identify a working solution. Seeing as you closed this issue yourself, I'm assuming that you solved it. Do you remember that solution?