nfcpy/nfcpy #148: documentation update: python 3 support

riesentoaster, 2019-12-09:
According to [this](https://github.com/nfcpy/nfcpy/issues/47#issuecomment-499693493) comment in an other issue, python 3 is supported. The [docs](https://nfcpy.readthedocs.io/en/latest/topics/get-started.html#py3issue) (footnote 1) say something else. 

Do the docs need an update?

nehpetsde, 2019-12-10:
Indeed the docs need an update. Thanks for reporting.

nehpetsde, 2019-12-10:
Documentation is updated. Thanks again for your help.

closed 2019-12-10 by nehpetsde

riesentoaster, 2019-12-11:
Thanks for updating it! Two more things:
- There is an other mistake [here](https://github.com/nfcpy/nfcpy/blob/master/docs/overview.rst#requirements).
- https://nfcpy.readthedocs.io/ is not yet updated. Does this happen automatically after some time?