nfcpy/nfcpy #150: Is it possible for me to change the polling interval with nfcpy?

sxiii, 2019-12-20:
Greetings!

I'm trying to shorten the polling interval of ISO 14443 Type A tag via the acr122u reader.

By default, I'm getting results of 615~715 msec operation. I'm approaching this as skiping the card types that I don't need while also turning off auto search etc.

So; I've realized that by default nfcpy sends this:
```
6f050000000000000000ff00517f00
```
According to documentation [here](http://downloads.acs.com.hk/drivers/en/API-ACR122U-2.02.pdf) - page 26 - I have to send either 10100001 (A1 in HEX) or 11100001 (E1 in HEX).

So, is there any easier option to send ```6f050000000000000000ff0051a100``` instead of default ```6f050000000000000000ff00517f00``` then rewriting functions of nfcpy?

Sorry for newbies questions and hope for your help.

sxiii, 2019-12-20:
I've seems to be getting it...
```
python3 sense.py 106A -r --atr 6f050000000000000000ff0051a100
```
That seems to shorten the polling interval for me.

nehpetsde, 2020-02-12:
This sounds like solved?

closed 2020-02-12 by nehpetsde

sxiii, 2020-02-19:
Yes, thank you @nehpetsde .