nfcpy/nfcpy #151: Not found device

papapepo, 2019-12-28:
Hello. I am trying to use nfcpy with my raspberry pi zero w and pn532 but after I install the library and check if it has been installed successfully I get a massage that no contactless device is found. What should I do?

nehpetsde, 2020-02-12:
Have you tried with `python -m nfc --search-tty`.

dupuisv, 2020-03-09:
try this

sudo sh -c 'echo SUBSYSTEM==\"usb\", ACTION==\"add\", ATTRS{idVendor}==\"04cc\", ATTRS{idProduct}==\"2533\", GROUP=\"plugdev\" >> /etc/udev/rules.d/nfcdev.rules'

then restart the pi

dupuisv, 2020-03-09:
remplace 04cc and 2533 by the results returned by lsusb, example:

lsusb
Bus 001 Device 006: ID 04cc:2533 ST-Ericsson NFC device (PN533)

TheScurvyDawg, 2021-11-01:
> 
> 
> try this
> 
> sudo sh -c 'echo SUBSYSTEM=="usb", ACTION=="add", ATTRS{idVendor}=="04cc", ATTRS{idProduct}=="2533", GROUP="plugdev" >> /etc/udev/rules.d/nfcdev.rules'
> 
> then restart the pi

I am sorry I need to ask but how can I reverse this should I try it?

dupuisv, 2021-11-04:
you can of course edit with
   sudo nano /etc/udev/rules.d/nfcdev.rules
to remove the line you added earlier.

dupuisv, 2021-11-04:
we use out own version of nfcpy that works with python3, but we didn't test all the functions, since we only read the cardnumber

nehpetsde, 2021-11-09:
Nfcpy should work well with Python3. Please let me know if it doesn't.