nfcpy/nfcpy #152: Emulation size: difference between tagtool.py and card emulation example in docs

riesentoaster, 2020-01-09:
So I'm trying to emulate a NFC tag on an SCL3711 (So PN533) that needs to provide a comparably big dataset (~3kbyte). If I use tagtool.py in conjunction with ndeftool, it works just fine:

```
ndeftool text "\[insert like 3000 chars\]" save --force ./text.ndef

./tagtool.py -l emulate ./text.ndef tt3
```

But when I use the example in the docs to card emulation (https://nfcpy.readthedocs.io/en/latest/topics/get-started.html#emulate-a-card), I can't read more than 3 blocks worth of ndef data. The text prior to the example states "The RC-S380 fully supports Type 3 Tag emulation. Devices based on PN532, PN533, or RC-S956 chipset can also be used but an internal frame size limit of 64 byte only allows read/write operations with up to 3 data blocks." 

Why is emulating bigger tags with the example not possible if it is with tagtool.py? Is this a fixable issue? Can I just tear apart tagtool.py if I need to emulate larger tags programmatically?

nehpetsde, 2020-02-12:
These two lines
```
ndef_data_area[1] = 12    # Number of blocks that may be read at once
ndef_data_area[2] = 8     # Number of blocks that may be written at once
```
control the maximum number of blocks (one block is 16 bytes) that the reader is allowed to read or write with a single command. Both values must not be more than 3 for a PN53x based frontend. In *tagtool.py* the default of those values is `1` and you can change that with `--nbr` and `--nbw` arguments placed after `tt3`.

riesentoaster, 2020-02-13:
I'm not sure I understand.

I have successfully emulated way more than the default 1 block using tagtool.py without adding additional arguments (like --nbr and --nbw). I just gave it a file with a larger ndef-message as input and it worked. 

So there either is no max for those values (--nbr/--nbw) and the emulation-example just contains a bug or there is a workaround in tagtool.py that hasn't made it into the example yet.

nehpetsde, 2020-02-13:
The emulation example uses values of nbr/nbw that can only be handled by the RC-S380 (these values are typical for FeluCa cards). If you want to emulate with a PN53x frontend you must change those to 3 or less.

The tagtool.py card emulation uses nbr/nbw of 1. That works with RC-S380 and PN53x.

Small values of nbr/nbr imply less communication speed because a smaller payload is transmitter per command.

The tag size (or NDEF data size) is a completely different thing.

riesentoaster, 2020-02-13:
Oh, so the max-read/write values don't need to be large enough to cover the entire ndef record?

nehpetsde, 2020-02-13:
No, these only dictate the maximum transfer unit per command. One block is 16 byte. Up to 12 blocks fit into a Type 3 Tag read command frame, 8 blocks into a write command frame. Large NDEF is transferred by multiple read or write commands, each addressing different blocks as needed.

riesentoaster, 2020-02-14:
I see. Thank you for the explanation!

closed 2020-02-14 by riesentoaster