nfcpy/nfcpy #154: Connection timed out after first connection to ACR122U-A9

gandy92, 2020-01-20:
I have just purchased a ACR122U-A9 which seems to work perfectly the first time after plugging it in. 

Running `python -m nfc` on a raspberry pi with raspbian stretch yields:
```
This is the 1.0.3 version of nfcpy run in Python 3.5.3
on Linux-4.14.79-v7+-armv7l-with-debian-9.6
I'm now searching your system for contactless devices
** found ACS ACR122U PN532v1.6 at usb:001:018
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
```

Afterwards, however, nfcpy seems to not find the reader again:
```
~/prg/nfc $ venv/bin/python -m nfc -vvv
This is the 1.0.3 version of nfcpy run in Python 3.5.3
on Linux-4.14.79-v7+-armv7l-with-debian-9.6
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.21
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:001:017
DEBUG:nfc.clf.transport:using libusb-1.0.21
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading acr122 driver for usb:072f:2200
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00480000
Level 9:nfc.clf.transport:<<< 800a000000000002810041435231323255323136
DEBUG:nfc.clf.acr122:initialize ACR122U216
DEBUG:nfc.clf.acr122:CCID ICC-POWER-ON
Level 9:nfc.clf.transport:>>> 62000000000000000000
Level 9:nfc.clf.transport:<<< 800200000000000081003b00
DEBUG:nfc.clf.acr122:Set PICC Operating Parameters
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00517f00
DEBUG:nfc.clf.device:[Errno 110] Connection timed out
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device
```
It is still listed in the usb subsystem, though:
```
~/prg/nfc $ lsusb
Bus 001 Device 018: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
Bus 001 Device 004: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
`dmesg` does not report anything after the usual device-just-got-plugged-in stuff:
```
[4845587.420638] usb 1-1.3: new full-speed USB device number 18 using dwc_otg
[4845587.562875] usb 1-1.3: New USB device found, idVendor=072f, idProduct=2200
[4845587.562890] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[4845587.562899] usb 1-1.3: Product: ACR122U PICC Interface
[4845587.562907] usb 1-1.3: Manufacturer: ACS
```

Only after pulling the USB plug and plugging the reader back in, it can be used once again before posing the same problem.

I set up a brand new virtual environment with fresh packages for the tests.

Is this a known problem and are there any workarounds? Would updating the distro help?

Thanks for any hints, cheers,
Andy.

gandy92, 2020-01-22:
Meanwhile, I set up a test environment on a Win7 system, with almost identical results:

First connection attempt after pluggin the reader in:
```
C:\prg\nfc\env-win\Scripts\python.exe -m nfc -vvv
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
This is the 1.0.3 version of nfcpy run in Python 3.7.3
on Windows-7-6.1.7601-SP1
I'm now searching your system for contactless devices
INFO:nfc.clf:searching for reader on path usb:003:010
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading acr122 driver for usb:072f:2200
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00480000
Level 9:nfc.clf.transport:<<< 800a000000000002810041435231323255323136
DEBUG:nfc.clf.acr122:initialize ACR122U216
DEBUG:nfc.clf.acr122:CCID ICC-POWER-ON
Level 9:nfc.clf.transport:>>> 62000000000000000000
Level 9:nfc.clf.transport:<<< 800200000000000081003b00
DEBUG:nfc.clf.acr122:Set PICC Operating Parameters
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00517f00
Level 9:nfc.clf.transport:<<< 80020000000000028100907f
DEBUG:nfc.clf.acr122:Configure Buzzer and LED
Level 9:nfc.clf.transport:>>> 6f090000000000000000ff00400e0400000000
Level 9:nfc.clf.transport:<<< 800200000000000281009002
Level 9:nfc.clf.acr122:Diagnose 00000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fa
Level 9:nfc.clf.transport:>>> 6f030100000000000000ff000000fed40000000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fa
Level 9:nfc.clf.transport:<<< 80000100000000028100d50100000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fa9000
Level 9:nfc.clf.acr122:GetFirmwareVersion 
Level 9:nfc.clf.transport:>>> 6f070000000000000000ff00000002d402
Level 9:nfc.clf.transport:<<< 80080000000000028100d503320106079000
DEBUG:nfc.clf.acr122:chipset is a PN532v1.6
Level 9:nfc.clf.acr122:SetParameters 00
Level 9:nfc.clf.transport:>>> 6f080000000000000000ff00000003d41200
Level 9:nfc.clf.transport:<<< 80040000000000028100d5139000
Level 9:nfc.clf.acr122:RFConfiguration 02000b0a
Level 9:nfc.clf.transport:>>> 6f0b0000000000000000ff00000006d43202000b0a
Level 9:nfc.clf.transport:<<< 80040000000000028100d5339000
Level 9:nfc.clf.acr122:RFConfiguration 0400
Level 9:nfc.clf.transport:>>> 6f090000000000000000ff00000004d4320400
Level 9:nfc.clf.transport:<<< 80040000000000028100d5339000
Level 9:nfc.clf.acr122:RFConfiguration 05010001
Level 9:nfc.clf.transport:>>> 6f0b0000000000000000ff00000006d43205010001
Level 9:nfc.clf.transport:<<< 80040000000000028100d5339000
DEBUG:nfc.clf.acr122:write analog settings for Type A 106 kbps
Level 9:nfc.clf.acr122:RFConfiguration 0a59f43f114d85616f266287
Level 9:nfc.clf.transport:>>> 6f130000000000000000ff0000000ed4320a59f43f114d85616f266287
Level 9:nfc.clf.transport:<<< 80040000000000028100d5339000
DEBUG:nfc.clf.acr122:write analog settings for Type F 212/424 kbps
Level 9:nfc.clf.acr122:RFConfiguration 0b69ff3f114185616f
Level 9:nfc.clf.transport:>>> 6f100000000000000000ff0000000bd4320b69ff3f114185616f
Level 9:nfc.clf.transport:<<< 80040000000000028100d5339000
DEBUG:nfc.clf.acr122:write analog settings for Type B 106 kbps
Level 9:nfc.clf.acr122:RFConfiguration 0cff0485
Level 9:nfc.clf.transport:>>> 6f0b0000000000000000ff00000006d4320cff0485
Level 9:nfc.clf.transport:<<< 80040000000000028100d5339000
DEBUG:nfc.clf.acr122:write analog settings for 14443-4 212/424/848 kbps
Level 9:nfc.clf.acr122:RFConfiguration 0d85158a8508b28501da
Level 9:nfc.clf.transport:>>> 6f110000000000000000ff0000000cd4320d85158a8508b28501da
Level 9:nfc.clf.transport:<<< 80040000000000028100d5339000
Level 9:nfc.clf.acr122:RFConfiguration 0102
Level 9:nfc.clf.transport:>>> 6f090000000000000000ff00000004d4320102
Level 9:nfc.clf.transport:<<< 80040000000000028100d5339000
INFO:nfc.clf:using ACS ACR122U PN532v1.6 at usb:003:010
Level 9:nfc.clf.transport:>>> 6f0600000000000000000000ff00ff00
Level 9:nfc.clf.transport:<<< 8000000000000002fe00
** found ACS ACR122U PN532v1.6 at usb:003:010
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
[... 13 lines cut ...]
[... cut ...]
Level 9:nfc.clf.transport:>>> 62000000000000000000
Level 9:nfc.clf.transport:<<< 800200000000000081003b00
DEBUG:nfc.clf.acr122:Set PICC Operating Parameters
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00517f00
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device
DEBUG:nfc.clf.device:[Errno 10060] Unknown error

Process finished with exit code 0
```

nehpetsde, 2020-01-22:
ACR122U213 was the latest hardware revision I could test. You have revision 216 and I remember problem reports already for revision 215 (see #105).

gandy92, 2020-02-02:
Thank you for the link to #105. I'll check if increasing the timeout works.

LiveDrunk, 2020-09-30:
Also looking for an update.
The reader i ordered was also revision 216.

GhobadMaleki, 2020-12-30:
Hi all, I also have the 216 revision. Any update on getting that working?

olavhaugen, 2021-01-23:
Exact same problem here. I'm trying to follow these instructions https://www.hackster.io/mark-hank/sonos-spotify-vinyl-emulator-3be63d but it fails to connect.

Is it not possible to get it to work?

kastrom, 2021-03-05:
I've now ordered two of these readers: one from Amazon UK and another from AliExpress. It's impossible to tell which revision you'll receive from the product listings and frustratingly both of mine are revision 216 which means I can't currently get them working

Anyone had any luck getting revision 216 to work?

gandy92, 2021-03-05:
Yes, got my rev 216 work like a charm using pcscd and pyscard

isspace112, 2021-07-28:
could you tell us how?

isspace112, 2021-07-29:
> Yes, got my rev 216 work like a charm using pcscd and pyscard

could you tell us how?

gandy92, 2021-07-29:
I started out with https://github.com/BrianHVB/intro-to-smartcard-development#heading--python-example-1 and it really worked out-of-the-box

Dennovin, 2022-01-02:
I got some of the non-working ones and after some experimentation, I think the timeout is too short in `set_buzzer_and_led_to_active`. With the defaults, it ends up being 0.4 seconds. If I send the "turn on for 0.3 seconds" command with a 0.4 second timeout, it fails, and no longer works until I unplug it. With 1 second, it works.

```
>>> from nfc.clf.acr122 import Chipset
>>> from nfc.clf import transport
>>> 
>>> found = transport.USB.find("usb:072f:2200")
>>> vid, pid, bus, dev = found[0]
>>> chipset = Chipset(transport.USB(bus, dev))
>>> 
>>> chipset.ccid_xfr_block(bytearray.fromhex("FF00400D0403000101"), timeout=1)
bytearray(b'\x90\x01')
>>> chipset.ccid_xfr_block(bytearray.fromhex("FF00400D0403000101"), timeout=0.4)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/nfc/clf/transport.py", line 313, in read
    frame = self.usb_dev.bulkRead(ep_addr, 300, timeout)
  File "/usr/local/lib/python3.9/dist-packages/usb1/__init__.py", line 1429, in bulkRead
    transferred = self._bulkTransfer(endpoint, data, length, timeout)
  File "/usr/local/lib/python3.9/dist-packages/usb1/__init__.py", line 1377, in _bulkTransfer
    mayRaiseUSBError(libusb1.libusb_bulk_transfer(
  File "/usr/local/lib/python3.9/dist-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/usr/local/lib/python3.9/dist-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorTimeout: LIBUSB_ERROR_TIMEOUT [-7]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/nfc/clf/acr122.py", line 210, in ccid_xfr_block
    frame = self.transport.read(int(timeout * 1000))
  File "/usr/local/lib/python3.9/dist-packages/nfc/clf/transport.py", line 315, in read
    raise IOError(errno.ETIMEDOUT, os.strerror(errno.ETIMEDOUT))
TimeoutError: [Errno 110] Connection timed out
```

Dennovin, 2022-01-13:
> Thanks @Dennovin . I think it's an implementation error that the command exchange time depends on the "turn on for .. seconds" time value. Can you confirm that with the 1 second command timeout other, longer, "turn on" values also work?

No, using a longer "turn on" value with a 1 second timeout breaks also.

```
>>> from nfc.clf.acr122 import Chipset
>>> from nfc.clf import transport
>>> found = transport.USB.find("usb:072f:2200")
>>> vid, pid, bus, dev = found[0]
>>> chipset = Chipset(transport.USB(bus, dev))
>>> chipset.ccid_xfr_block(bytearray.fromhex("FF00400D040A000101"), timeout=1)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/nfc/clf/transport.py", line 313, in read
    frame = self.usb_dev.bulkRead(ep_addr, 300, timeout)
  File "/usr/local/lib/python3.9/dist-packages/usb1/__init__.py", line 1429, in bulkRead
    transferred = self._bulkTransfer(endpoint, data, length, timeout)
  File "/usr/local/lib/python3.9/dist-packages/usb1/__init__.py", line 1377, in _bulkTransfer
    mayRaiseUSBError(libusb1.libusb_bulk_transfer(
  File "/usr/local/lib/python3.9/dist-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/usr/local/lib/python3.9/dist-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorTimeout: LIBUSB_ERROR_TIMEOUT [-7]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/nfc/clf/acr122.py", line 210, in ccid_xfr_block
    frame = self.transport.read(int(timeout * 1000))
  File "/usr/local/lib/python3.9/dist-packages/nfc/clf/transport.py", line 315, in read
    raise IOError(errno.ETIMEDOUT, os.strerror(errno.ETIMEDOUT))
TimeoutError: [Errno 110] Connection timed out
```

Mallon94, 2022-02-14:
Hi guys, 

For those stuck with this error, I have found a work around.
When installing LibUSB using Zadig, install libusbK and not WinUSB.

Then, when we have got a read, end your script by using this
<subprocess.run(['pnputil /restart-device "USB\VID_072F&PID_2200\6&B071D68&0&1"'])>

This uses windows PnPutil command to restart the ACR122u device, which frees up the device and allows for a new connection.

Hope this helps!

fabiom84, 2022-02-18:
Dear @Mallon94 thanks for your suggestion, but if I change driver I am then unable to connect to device with nfcpy (under windows 10) I get error when i do clf=nfc.ContactlessFrontend('usb'):
 raise IOError(errno.ENODEV, os.strerror(errno.ENODEV))
OSError: [Errno 19] No such device

it seems I need to stay with winusb driver..

Any idea?

thanks!

Mallon94, 2022-02-18:
Did you get the device instance ID from device manager for your reader?

The Errno 19 No such device error is typical of the reader when it hangs unresponsive. If you go into an elevated(run as admin) cmd and run this command

fabiom84, 2022-02-18:
Dear @Mallon94 how can I run a command as elevated from inside python script? I really have not knowledge in coding, maybe it's a silly question but I do not know how to do it.. thanks