nfcpy/nfcpy #157: Adafruit PN532 V1.6 + FTDI TTL-232R-3V3 not working

mitea1, 2020-02-14:
Hi

I'm using Adafruits PN532 Breakoutboard which is connected via FTDI TTL-232R-3V3 cable to my Win 10 machine.
Jumpers SEL0 and SEL1 are connected to OFF. LED is on so device seems to run.

I did every step to described in the Getting started section. https://nfcpy.readthedocs.io/en/latest/topics/get-started.html

- Using Zadiq I selected the FTDI device and updated its driver to WinUSB(v6.1.7600.16385)
- downloaded libusb 1.0.22 and copied the files as described 
- restarted my machine 

when i run `python -m nfc` I get:

`This is the 1.0.3 version of nfcpy run in Python 3.7.4
on Windows-10-10.0.17763-SP0
I'm now searching your system for contactless devices
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device
`
 `python -m nfc --search-tty` doesn't help either
what am I doing wrong?

nehpetsde, 2020-02-14:
The driver installation is only needs for devices that communicate as USB devices. What you have is a serial communication tunneled over USB. This is handled by the FTDI driver and made accessible as virtual COM ports.

mitea1, 2020-02-14:
thanks a lot for your quick reply!

I'm using the standard FTDI driver now and the device is again shown in the Device Manager under Ports as TTL232R-3V3 (COM22)

I tried this one: 

```
import nfc
clf = nfc.ContactlessFrontend()
clf.open("com:COM22:pn532")
```

but this results in an error

`    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
`

what am I doing wrong? Is my code snipped correct at all?

mitea1, 2020-02-18:
It works now. Thanks for your support! 

Problem was due to my driver updates my FTDI didn't use the correct drivers in windows anymore. couldn't even use the FTDI in my terminal tool. had to manually select one in device manager.

maybe you could update the documentation regarding this to have bulletproof instructions?

closed 2020-02-18 by mitea1