nfcpy/nfcpy #160: Import Error

vietta90, 2020-03-28:
I am having an issue importing nfc and am getting an error stating that message_decoder cannot be imported similar to a previous issue with the same error message. https://github.com/nfcpy/nfcpy/issues/88
I am using raspbian on a raspberry pi 4 with python 2.7.
I have ndeflib installed as suggested by one of the suggestion. I would appreciate any help.

nehpetsde, 2020-03-29:
ndeflib should be automatically installed as a dependency of nfcpy, let me know if that was not the case. To verify installation, you should create and activate a Python virtual environment.
```shell
$ virtualenv -p python2 my-venv
$ . my-venv/bin/activate
$ pip install nfcpy
```

vietta90, 2020-03-29:
Thank for the help I was able to get it to work. It was due to using sudo when running the python.

closed 2020-03-29 by vietta90