nfcpy/nfcpy #167: Error: Failed to retrieve ACR122U version string/ insufficient data for decoding chip response

nemesis90, 2020-07-01:
Hello,

I get the following errors when I try to connect to the ACR122u, any suggestions?

Results ------------------

One:

python -m nfc
This is the 1.0.3 version of nfcpy run in Python 2.7.16
on Linux-4.19.118-v7l+-armv7l-with-debian-10.4
I'm now searching your system for contactless devices
ERROR:nfc.clf.acr122:failed to retrieve ACR122U version string
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device

Two:

python -m nfc
This is the 1.0.3 version of nfcpy run in Python 2.7.16
on Linux-4.19.118-v7l+-armv7l-with-debian-10.4
I'm now searching your system for contactless devices
ERROR:nfc.clf.acr122:insufficient data for decoding chip response
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device

Any suggestions?

nehpetsde, 2020-07-01:
Add the debug flag -d and post the output of the first run after USB plug-in.

nemesis90, 2020-07-01:
> Add the debug flag -d and post the output of the first run after USB plug-in.

I was able to add the --verbose option. How do add the -d option?

python -m nfc --verbose
This is the 1.0.3 version of nfcpy run in Python 2.7.16
on Linux-4.19.118-v7l+-armv7l-with-debian-10.4
I'm now searching your system for contactless devices
INFO:nfc.clf:searching for reader on path usb:001:005
ERROR:nfc.clf.acr122:failed to retrieve ACR122U version string
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device

nemesis90, 2020-07-02:
Here is more info using some of the documentation calls: ( assert clf.open(...) )

[image]
[image]

nehpetsde, 2020-07-02:
Sorry, debug output comes with multiple -v options. Give it four times for the greatest verbosity.

nemesis90, 2020-07-02:
Here is the debug output. Thanks for your help on this. 

python -m nfc -v -v -v -v
This is the 1.0.3 version of nfcpy run in Python 2.7.16
on Linux-4.19.118-v7l+-armv7l-with-debian-10.4
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:001:009
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading acr122 driver for usb:072f:2200
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00480000
Level 9:nfc.clf.transport:<<< 800a000000000002810041435231323255323135
DEBUG:nfc.clf.acr122:initialize ACR122U215
DEBUG:nfc.clf.acr122:CCID ICC-POWER-ON
Level 9:nfc.clf.transport:>>> 62000000000000000000
Level 9:nfc.clf.transport:<<< 800200000000000081003b00
DEBUG:nfc.clf.acr122:Set PICC Operating Parameters
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00517f00
Level 9:nfc.clf.transport:<<< 80020000000000028100907f
DEBUG:nfc.clf.acr122:Configure Buzzer and LED
Level 9:nfc.clf.transport:>>> 6f090000000000000000ff00400e0400000000
Level 9:nfc.clf.transport:<<< 800200000000000281009002
Level 9:nfc.clf.acr122:Diagnose 00000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fa
Level 9:nfc.clf.transport:>>> 6f030100000000000000ff000000fed40000000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2f3f4f5f6f7f8f9fa
Level 9:nfc.clf.transport:<<< 8000000000000002fe00
ERROR:nfc.clf.acr122:insufficient data for decoding chip response
DEBUG:nfc.clf.device:[Errno 5] Input/output error
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device

nemesis90, 2020-07-02:
I ran it a second time and here is what I got:

 python -m nfc -v -v -v -v
This is the 1.0.3 version of nfcpy run in Python 2.7.16
on Linux-4.19.118-v7l+-armv7l-with-debian-10.4
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:001:009
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading acr122 driver for usb:072f:2200
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00480000
Level 9:nfc.clf.transport:<<< 800a000000000000810041435231323255323135
DEBUG:nfc.clf.acr122:initialize ACR122U215
DEBUG:nfc.clf.acr122:CCID ICC-POWER-ON
Level 9:nfc.clf.transport:>>> 62000000000000000000
Level 9:nfc.clf.transport:<<< 800a000000000000810041435231323255323135
DEBUG:nfc.clf.acr122:Set PICC Operating Parameters
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00517f00
Level 9:nfc.clf.transport:<<< 800200000000000081003b00
DEBUG:nfc.clf.acr122:Configure Buzzer and LED
Level 9:nfc.clf.transport:>>> 6f090000000000000000ff00400e0400000000
DEBUG:nfc.clf.device:[Errno 110] Connection timed out
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device

nemesis90, 2020-07-02:
Running it with only three verbose gives the following:

python -m nfc -v -v -v 
This is the 1.0.3 version of nfcpy run in Python 2.7.16
on Linux-4.19.118-v7l+-armv7l-with-debian-10.4
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:001:009
DEBUG:nfc.clf.transport:using libusb-1.0.22
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading acr122 driver for usb:072f:2200
Level 9:nfc.clf.transport:>>> 6f050000000000000000ff00480000
Level 9:nfc.clf.transport:<<< 80020000000000008100907f
ERROR:nfc.clf.acr122:failed to retrieve ACR122U version string
DEBUG:nfc.clf.device:[Errno 19] No such device
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device

nehpetsde, 2020-07-07:
You are having an ACR122U revision 215 and unfortunately this seems not be usable with nfcpy. See #105 for that same model and #154 for a revision 216 device. as stated there, I can not help further without being able to test with the device. Feel free to submit a patch if you get it working.

closed 2020-09-22 by nehpetsde

LiveDrunk, 2020-09-30:
I have the same issue. I have revision 216...
Any thoughts on how this can be fixed?
Or any way on how i can check if there is another program keeping is from connecting properly?

LiveDrunk, 2020-09-30:
Try to run the application the first time without checking with: `python -m nfc`
And try to keep it running so the connection doesn't go bad.