nfcpy/nfcpy #175: TypeError: a bytes-like object is required, not 'str'

bkupidura, 2020-09-11:
Looks like tag.authenticate() for tt2_nxp is broken.

```
  File "/usr/local/lib/python3.7/dist-packages/nfc/tag/tt2_nxp.py", line 467, in authenticate
    return super(NTAG21x, self).authenticate(password)
  File "/usr/local/lib/python3.7/dist-packages/nfc/tag/__init__.py", line 367, in authenticate
    self._authenticated = self._authenticate(password)
  File "/usr/local/lib/python3.7/dist-packages/nfc/tag/tt2_nxp.py", line 474, in _authenticate
    log.debug("authenticate with key %s", hexlify(key).decode())
TypeError: a bytes-like object is required, not 'str'
```

Python 3.7.3
nfcpy 1.0.3

nehpetsde, 2020-09-21:
The password must be given as a byte sequence (ultimately a password is a sequence of bits with no relation to the concept of words). Python 3 strings are all unicode and the library can not take responsibility for converting into a byte representation - of which there would be many. This is up to library user.

nolansingroy, 2021-03-13:
> The password must be given as a byte sequence (ultimately a password is a sequence of bits with no relation to the concept of words). Python 3 strings are all unicode and the library can not take [...]

What's a good 3rd party library to use that you would recommend to convert a string to bytes? I came across this issue as well What's a good 3rd party library to use that you would recommend to convert a string to bytes? I came across this issue as well.
 
[image]

markfilan, 2021-12-23:
The reason for this error is that in Python 3, strings are Unicode, but when transmitting on the network, the data needs to be bytes instead. We can convert [bytes to string](http://net-informations.com/python/iq/byte.htm) using bytes class decode() instance method, So you need to decode the bytes object to produce a string. In Python 3 , the default encoding is "utf-8" , so you can use directly:
CODE: SELECT ALL

`b"python byte to string".decode("utf-8")`

Python makes a clear distinction between bytes and strings . Bytes objects contain raw data — a sequence of octets — whereas strings are Unicode sequences . Conversion between these two types is explicit: you encode a string to get bytes, specifying an encoding (which defaults to UTF-8); and you decode bytes to get a string. Clients of these functions should be aware that such conversions may fail, and should consider how failures are handled.