nfcpy/nfcpy #176: MFRC522 support

socram8888, 2020-09-17:
Hello. I wrote a while ago a tiny Python class for dealing with MFRC522 devices connected via an USB UART device with a PC (see https://github.com/socram8888/py522/blob/master/py522/reader/rc522.py)

I think it could be useful to have it integrated into this project, and thus allow using from Python easily a sub-2$ device that, although ISO14443A-only, would be capable of dealing with most of the common tags.

If you think it's okay and useful I'd work on integrating it myself.

nehpetsde, 2020-09-21:
Many thanks for the proposal. I think it will be useful and certainly appreciate the contribution.

Ovcojedac, 2022-07-20:
Hey @nehpetsde has this ever entered the nfcpy? Would be happy to contribute to integration and testing :)

Ovcojedac, 2022-07-23:
@socram8888 maybe you have some updates?

socram8888, 2022-07-23:
None, sorry. I ended up acquiring a Proxmark