nfcpy/nfcpy #177: Does nfcpy support i2c connection of PN532 board with Raspberry Pi

ghxbob, 2020-09-18:
Hi,
I saw there is only description of uart connection of PN532 board with Raspberry, 
Does nfcpy support i2c connection of PN532 board with Raspberry Pi?

I've already test the i2c connection of PN532 board with Raspberry Pi, however i got following error
ERROR:nfc.clf:no reader available on path tty:S0

pi@raspberrypi:~ $ python -m nfc --search-tty
This is the 1.0.3 version of nfcpy run in Python 2.7.16
on Linux-5.4.51-v7l+-armv7l-with-debian-10.4
I'm now searching your system for contactless devices
ERROR:nfc.clf:no reader available on path tty:S0

Results of testing i2c connection with libnfc
pi@raspberrypi:~ $ nfc-poll
nfc-poll uses libnfc 1.8.0
NFC reader: _PN532_I2c opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  08  
       UID (NFCID1): 8f  dc  08  ef  
      SAK (SEL_RES): 28  
                ATS: 77  f7  a0  02  47  44  
Waiting for card removing...

ghxbob, 2020-09-18:
more information:

ls -l /dev/tty[ASU]* /dev/serial?
lrwxrwxrwx 1 root root          5 Sep 18 14:33 /dev/serial0 -> ttyS0
lrwxrwxrwx 1 root root          7 Sep 18 14:33 /dev/serial1 -> ttyAMA0
crw-rw---- 1 root dialout 204, 64 Sep 18 14:33 /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 65 Sep 18 14:33 /dev/ttyAMA1
crw-rw---- 1 root dialout 204, 66 Sep 18 14:33 /dev/ttyAMA2
crw-rw---- 1 root dialout 204, 67 Sep 18 14:33 /dev/ttyAMA3
crw-rw---- 1 root dialout 204, 68 Sep 18 14:33 /dev/ttyAMA4
crw-rw---- 1 root dialout   4, 64 Sep 18 14:45 /dev/ttyS0

nehpetsde, 2020-09-21:
There is no support for I2C reader connection. I simply never found the time and it wasn't contributed by anyone else.

ghxbob, 2020-09-21:
after   $ ./configure --with-drivers=pn532_uart --enable-serial-autoprobe，UART connection of PN532 and Raspberry Pi 4B works now, see more info 
https://learn.adafruit.com/adafruit-pn532-rfid-nfc/libnfc

ghxbob, 2020-09-22:
Hi 
I'd like to contribute on i2c connection,  however I have no such experience,  any guide or reference code?

ghxbob, 2020-10-05:
@nehpetsde Your comment is appreated, any advice on i2c reader connection?

closed 2021-01-20 by ghxbob