nfcpy/nfcpy #186: PN532: Powerdown command on shutdown prevents further communication

matthijskooijman, 2021-03-12:
I'm using this library with a PN532 device, communicating over HSU. However, when I properly close down the library, using `clf.close()`, I can no longer start communication afterwards (I get a communication timeout).

I highly suspect that this might be a firmware issue in the PN532 chip, but I wondered if anyone else has been seeing the same.

The problem seems to be caused by the powerdown command that is sent as the last command here: https://github.com/nfcpy/nfcpy/blob/489ba9d9fe974e640cd84fe04f688090bd6f5c6b/src/nfc/clf/pn532.py#L232

It does specify that it should allow wakeup on HSU, but it seems it does not. When I remove that powerdown command, everything works as expected. I tried removing all other code from that close function (i.e. the extra ACK, delay and baudrate change), leaving just the powerdown and I also tried replacing that other code with a 1 second delay (after reading [this post of someone having a problem when powerdown was issued shortly after another command](https://www.avrfreaks.net/forum/pn532-nfc-wake-i2c-doesnt-work-if-target-powers-down-too-early)), but in both cases the problems persisted.

I also checked the UART lines using a logic analyzer, and confirmed that the communication looks as expected: The last command is the powerdown command, with the right bits set, which is confirmed with a response with status=0. Then, trying to restart my application, I see the get version command with a bunch of leading zeroes as a preamble transmitted correctly, but no reply is sent.

Note that I've modified the library slightly to not change the baudrate (to make it easier to restart when communication is interrupted), but I do not believe this is relevant here.

So, now I wonder:
 - Are others seeing the same issue with the PN532? Or is this working for everyone else?
 - If this is indeed broken in the firmware, should nfcpy maybe not send the powerdown command? Or maybe make it optional at least?

matthijskooijman, 2021-03-12:
I also tried changing the powerdown command to:

     self.chipset.power_down(wakeup_enable=("HSU",)) 

Just in case wakeup on the other sources somehow confused the firmware, but that did not help either.

jeremielalanne, 2024-01-04:
Hello!
Might help others, but I found a very simple workaround that works in this case, just trying again haha. It's in .NET but same process can be translated for python, I guess they have same functions.

```
     public void PowerDown() {
         while(_Pn532.PowerDown(WakeUpEnable.I2c)) {
             Console.WriteLine("Could not power down PN532. Trying again...");
             Thread.Sleep(500);
         }
         Console.WriteLine("PN532 successfully powered down.");
     }
```

matthijskooijman, 2024-01-04:
@jeremielalanne Thanks for your comment. But my issue was that the powerdown did (I guess - cannot remember if there was some kind of ACK or not) succesful, it's just that it would not respond to further commands after that - i.e. it would not wake up. Or are you seeing the same if you try powerdown only once and that command is not succesful? And then trying the powerdown again until it does get an ACK does allow it to powerup again?

jeremielalanne, 2024-01-04:
> @jeremielalanne Thanks for your comment. But my issue was that the powerdown did (I guess - cannot remember if there was some kind of ACK or not) succesful, it's just that it would not respond to [...]

Yes I understand your issue, I was having the same :) When for example I would soft reboot, or restart the app, I couldn't communicate with it anymore, it wouldn't respond, because the communication did not end correctly at first. But adding a retry process made it work. It almost always failed at first try, but now it succeeds at second. i tried to restart my app few times, soft reboot, worked each time :) it surely isn't a perfect solution, but a workaround. Some kind of cancellation token should be added to it to avoid infinite loop.

bofh69, 2024-03-09:
I have a similar issue with a Elechouse PN532 V3 connected via UART to a Raspberry Pi 3B with buster. It stops responding after I stop my program.

For me it didn't help to disabling the power_down call though. I've not yet tried your other change with the serial port's speed.

I didn't notice this when I had it connected to another pi 3B (with bullseye), so it could be a different issue. That pi just had the nfc reader connected and short cables. The one where I have it now has SPI active with an external LCD, a few IOs, two USB serial ports and longer cables to the PN532. On the other hand it starts working if I power cycle the PN532, so it doesn't really seem to be a problem with the pi. Maybe the different python versions matter though.

bofh69, 2024-03-09:
Now I've tested some more.

I have two programs. One that doesn't call clf.close() , one that does.
If I ran either of the programs, I couldn't get either to work again without power cycling the PN532.

Just disabling the change of serial port speed fixed it for me though. Now that I think about it, the other pi used the mini-uart, where change_baudrate will be set to False anyway.

bofh69, 2024-03-11:
There could be a bug in my program so the clf.close() is never called. I'll verify that it is called and report back.

bofh69, 2024-03-13:
clf.close() is called.

It is possible to get the firmware version from the chip by sending 0xD4 0x02.
The returned frame for me is 0xD5 0x03 0x32 0x01 0x06 0x07; meaning version 1, revision 6.

matthijskooijman, 2024-04-09:
> clf.close() is called.
> 
> It is possible to get the firmware version from the chip by sending 0xD4 0x02. The returned frame for me is 0xD5 0x03 0x32 0x01 0x06 0x07; meaning version 1, revision 6.

I've read your comment and I am planning to check my PN532 chips for firmware version, but haven't gotten around to it yet. But it's still on my mind :-)

matthijskooijman, 2024-05-14:
I finally got around to do some more testing, and the two boards I tested both seem to handle the powerdown command just fine. They also have the same firmware version as shown by @bofh69. I *think* the original board I was seeing this problem with might be an elechouse v3 board and I suspect that particular board is in one of our production readers. I can see if I can swap it out sometime soon and also test that board.

I've now tested these two boards:
 - Clone board (probably ordered from https://www.tinytronics.nl/en/communication-and-signals/wireless/rfid/rfid-nfc-kit-pn532-with-s50-card-and-s50-key-tag (photos on site do not match - my board does not say "ELECHOUSE", just "NFC MODULE V3" and ".HW"). Given the price, I suspect this might have a clone PN532 chip as well, I've heard these exist. firmware version: b'32010607'. Power down works as expected.
 - Genuine elechouse v4 board (ordered directly from them). firmware version: b'32010607'. Powerdown works as expected.

This is the code I used to get the firmware version:
```diff
--- a/examples/sense.py
+++ b/examples/sense.py
@@ -48,6 +48,9 @@ def main(args):
 
     clf = nfc.ContactlessFrontend()
     if clf.open(args.device):
+        import binascii
+        version = clf.device.chipset.get_firmware_version()
+        print("firmware version: {}".format(binascii.hexlify(version)))
         targets = list()
         for target in args.targets:
             target_pattern_match = target_pattern.match(target)
```

    $ python3 examples/sense.py --device ttyACM0:pn532 -r -v 106A 106B 212F 424F

matthijskooijman, 2024-06-18:
Today I tested the last card reader we have (probably also the on that I used when creating this issue), and that indeed shows the powerdown problem. If I leave powerdown enabled, I can no longer connect a second time, the reader just does not respond. The reported firmware version is the same as the other ones: `firmware version: b'32010607'`.

Not that I tested this reader while it was still installed (connected to the hardware serial port of a beaglebone black), while my other tests were testing just the card reader (connected to an USB serial device), so that might also be a cause of this difference, but I expect it is not.

I also have not opened up the card reader yet to do a physical examination yet.