nfcpy/nfcpy #194: Read NDEF / nfc tags reference compatible ?

epsilon3379, 2021-07-22:
Hi !

I have the usb nfc reader ACR122U. I also have some type 2 Ntag215 cards [https://www.amazon.fr/gp/product/B06Y4PJSPN/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](url)

I have wrote some ndef entries with my smatphone and the nxp app (tag writer) with no problem. I can read the entries with the NFC tool pro app too.

But I can't when I try with nfcpy. Here's my code :

`import json
import nfc
import ndef

clf = nfc.ContactlessFrontend('usb')
from nfc.clf import RemoteTarget

print(clf)

target = clf.sense(RemoteTarget('106A'))

if target is not None:
    print(target)
    tag = nfc.tag.activate(clf, target)
    print(tag.dump)

    toast = [ndef.TextRecord("test 001")]

    if tag.ndef is not None:
        print(tag.ndef)
    else:
        print("NDEF data not present.")
        
else:
    print("Tag is not present.")

clf.close()`

And this is the output console :

ACS ACR122U on usb:001:005
106A sdd_res=333942E3 sel_res=08 sens_res=0400
<bound method Type2Tag.dump of <nfc.tag.tt2.Type2Tag object at 0x000001F2168A6F10>>
NDEF data not present.

What's the problem ?  I can't read the ndef entries with this code. I've read that some nfc tag can't be use with nfcpy because of encryption problem or something like this..

If it's the problem, which reference of nfc card can I use to make this working ?

Thanks :)

grandcat, 2021-07-23:
I encountered the same problem reading a TAG4A tag via `tty:USB0:pn532`. It seems the library cannot detect the presence of `ndef` records. I had not a chance yet to dig deeper into this library though.

Thanks to the author of this library at this point, it is pretty nice :+1:

epsilon3379, 2021-07-23:
Then I'm stuck...

In fact, I just want to write a string into my card and read it, no matter if its a ndef record or not. I don't know if it's mandatory to store datas into ndef records or something.

How can I simply read and write a string into my nfc tag then ?

Thanks ;)

grandcat, 2021-07-25:
I just realized that there is an unreleased fix on master which makes ndef records for T4 tags work.
Just try it with current master.

epsilon3379, 2021-07-25:
My tag is a Type 2, not Type 4.. 
I have installed nfcpy with the "pip install -U nfcpy" so I already have the latest vesion of nfcpy.

All I need to know is: 

- How can I read/write on my nfc tag ? (if its possible) or,
- Anybody know some tags references which works with ndef records through nfcpy ??

Thanks ;)

grandcat, 2021-07-25:
`pip install -U nfcpy` will install the latest release version, but not the master version. I would give master of this repo a try.

epsilon3379, 2021-07-25:
Just tested with the master branch of the repo, same result... no NDEF records can be read..

nehpetsde, 2021-07-26:
You can try if the `examples/tagtool.py` script can read your tag and you could activate debug logging to check further.

And I see you explicitly importing `ndef` - make sure you don't have the https://pypi.org/project/ndef module installed, the one required by `nfcpy` is https://pypi.org/project/ndeflib and is automatically installed (use `pip list` to find out).

epsilon3379, 2021-08-01:
The tagtool.py give me this output:

`[nfc.clf] searching for reader on path usb
[nfc.clf] using ACS ACR122U PN532v1.6 at usb:001:004
** waiting for a tag **
Type2Tag ID=333942E3`

I already know that my tag is an Type2tag, I don't know how it can helps 🤔

nehpetsde, 2021-08-19:
Please run tagtool.py with the debug flag `-d nfc` and post the output here.

epsilon3379, 2021-08-19:
Sure:

`
[main] enable debug output for 'nfc'
[nfc.clf] searching for reader on path usb 
[nfc.clf.transport] using libusb-1.0.24    
[nfc.clf.transport] path matches '^(usb|)$'
[nfc.clf.device] loading acr122 driver for usb:072f:2200
[nfc.clf.acr122] initialize ACR122U214
[nfc.clf.acr122] CCID ICC-POWER-ON
[nfc.clf.acr122] Set PICC Operating Parameters
[nfc.clf.acr122] Configure Buzzer and LED     
[nfc.clf.acr122] chipset is a PN532v1.6
[nfc.clf.acr122] write analog settings for Type A 106 kbps
[nfc.clf.acr122] write analog settings for Type F 212/424 kbps
[nfc.clf.acr122] write analog settings for Type B 106 kbps
[nfc.clf.acr122] write analog settings for 14443-4 212/424/848 kbps
[nfc.clf] using ACS ACR122U PN532v1.6 at usb:001:012
[nfc.clf] connect('rdwr', 'card')
** waiting for a tag **
[nfc.clf] removing card_options after on-startup
[nfc.clf] connect options after startup: rdwr
[nfc.clf] sense 106A 
[nfc.clf.acr122] disable crc check for type 2 tag
[nfc.clf] found 106A sdd_res=333942E3 sel_res=08 sens_res=0400
[nfc.clf] found 106A sdd_res=333942E3 sel_res=08 sens_res=0400
[nfc.clf] discovered target 106A sdd_res=333942E3 sel_res=08 sens_res=0400
[nfc.tag] trying to activate 106A sdd_res=333942E3 sel_res=08 sens_res=0400
[nfc.tag] trying type 2 tag activation for 106A
[nfc.clf] connected to Type2Tag ID=333942E3
Type2Tag ID=333942E3
[nfc.tag.tt2] read ndef data
[nfc.tag.tt2] read pages 0 to 3
[nfc.tag.tt2] >> 3000 (0.005000s)
[nfc.clf] >>> 3000 timeout=0.005
[nfc.clf] <<< 04 0.021s
[nfc.tag.tt2] << 04 (0.021928s)
[nfc.tag.tt2] received nak response
[nfc.clf] sense 106A sdd_res=333942E3 sel_req=333942E3 sel_res=08 sens_res=0400
[nfc.clf.acr122] disable crc check for type 2 tag
[nfc.clf] found 106A sdd_res=333942E3 sel_res=08 sens_res=0400
[nfc.clf] found 106A sdd_res=333942E3 sel_res=08 sens_res=0400
[nfc.tag.tt2] first four memory pages were unreadable
`

nehpetsde, 2021-08-20:
It looks like this is an NTAG not manufactured by NXP but the AMIC Technology Group (according to the chip manufacturer ID). It should still be recognized as a standard Type 2 Tag but returns an error response when trying to read the first page (16 bytes). It may possibly return data when the read is retried - maybe you could try that.

epsilon3379, 2021-08-20:
Retry the read ? How can I do that ?
By launching tagtool twice ?

Alternatively, do you know some NFC tag references which work 100% with nfcpy ?

nehpetsde, 2021-08-20:
Retrying the read would require you to program it and buying other tags may be a more pragmatic solution. All NTAGs I've seen were manufactured by NXP and work flawless. Unfortunately they often look just the same - one visually distinguishable type is this https://www.amazon.de/NFC21-68347-Anh%C3%A4nger-Style-Farben/dp/B00K5SM8YS/ref=mp_s_a_1_5?dchild=1&keywords=ntag&qid=1629483726&sr=8-5