nfcpy/nfcpy #196: import error

antwi23, 2021-08-19:
Hey guys, maybe you can help  me.
I got some problems with beam.py

1.)
Without sudo, "python3 beam.py send text TEXT --lang STRING",i get following error description: 

Traceback (most recent call last):
...
...
...
ModuleNotFoundError: No module named 'serial.tools'.

2.)
When i write  "sudo python3 beam.py send text TEXT --lang STRING" ,i get following error description:
Traceback (most recent call last):
  File "beam.py", line 29, in <module>
    import ndef
ModuleNotFoundError: No module named 'ndef'

but for example when i try to install ndef i get following  message : 

Requirement already satisfied: ndef in /home/pi/.local/lib/python3.7/site-packages (0.2).

So why i got different types of error description because of sudo and why  don't find my pi  ndef and/or serial.tools and how i could fix this ??

nehpetsde, 2021-08-20:
I'd recommend to try first within a Python virtual environment:
```
python -m venv venv
source venv/bin/activate
pip install nfcpy
```

antwi23, 2021-08-30:
thanks, now it is working.

closed 2021-08-30 by antwi23