nfcpy/nfcpy #203: Reader fails to open unless properly closed (ERRNO 110)

crespum, 2021-11-09:
Hello,

I'm using a PN5321 connected through an FTDI on Linux.

The first time everything works as a charm, however, if the program is stopped without calling `close()` method (for instance if it crashes), I can never open the reader again unless I manually reset it. This behavior doesn't happen if I close the port properly.

This is the error I'm seeing:
```
Traceback (most recent call last):
  File "src/nfc_reader.py", line 32, in __init__
    assert self.pn532.open(serialport) is True
  File ".venv/lib64/python3.10/site-packages/nfc/clf/__init__.py", line 149, in open
    self.device = device.connect(path)
  File ".venv/lib64/python3.10/site-packages/nfc/clf/device.py", line 112, in connect
    device = driver.init(tty)
  File ".venv/lib64/python3.10/site-packages/nfc/clf/pn532.py", line 411, in init
    if not transport.read(timeout=initial_timeout) == Chipset.ACK:
  File ".venv/lib64/python3.10/site-packages/nfc/clf/transport.py", line 153, in read
    raise IOError(errno.ETIMEDOUT, os.strerror(errno.ETIMEDOUT))
TimeoutError: [Errno 110] Connection timed out
```

Is there any way to overcome this issue?

nehpetsde, 2021-11-09:
Not to my knowledge. The only thing that might enable programmatic reset is to toggle USB power. But this is out of scope for the library.

crespum, 2021-11-11:
I don't think I've experienced this issue with other libraries I was using. I'm going to try it over and let you know.

Thanks for the prompt reply.

nehpetsde, 2021-11-11:
A simple architectural difference might be that other libraries might be using a separate driver process (like the Linux NFC subsystem, or a PC/SC stack).