nfcpy/nfcpy #206: Question about read/write notifications

eriknyquist, 2021-12-15:
Hello,

I have an embedded target emulating a Type2 tag, and I am using your Type2TagMemoryReader class to read/write a memory location on that tag.

I'm trying to figure out some way to be notified whenever any part of the memory is written to / changed by the tag-- so that if my embedded target firmware writes new data to the tag memory, my nfcpy script can instantly read the new data, without having to poll and check by reading the same location repeatedly.

I know that LLCP and SNEP protocols provide some level of abstraction that might get me closer to what I want, but in this case I am trying to do it without requiring LLCP or SNEP to be implemented on the embedded target side.

Is this kind of thing possible with nfcpy? Or, perhaps you could point me in the right direction if I was to make some modifications to achieve this?

nehpetsde, 2021-12-17:
The reader-tag communication is strictly such that the reader sends commands and the tag responds, the target can never actively send data. If it is the tag that has something to say then the reader must periodically ask - no way around it. If you are looking for something that gives the appearance of spontaneous tag-initiated communication then you may have a look at the PHDC TagManager / TagAgent code in `examples/`.

eriknyquist, 2021-12-18:
Thank you for the clarification and the pointer to example code, @nehpetsde ! much appreciated.

closed 2021-12-18 by eriknyquist