nfcpy/nfcpy #209: Does nfcpy support Type 5 NFC tags?

Fresbee, 2022-02-21:
I have a Type 5 NFC tag, which uses ISO/IEC 15693 communication protocols.  I wasn't sure if nfcpy supports this newer type of tag.  The documentation only lists Tag types 1-4.  If this module supports NFC Tag 5, what settings would I use to try reading and writing to it?  

I am using a Sony RC-S380/S NFC reader device.  I am able to read Type 2 and Type 3 NFC tags with it, but cannot detect the NFC Type 5 tag (ST25DV module).  The Type 5 tag is working, since my Android phone can read and write data to it.

nehpetsde, 2022-02-23:
I'm sorry, Type 5 tags are not supported.

Fresbee, 2022-02-24:
I see.  Would this be a possible addition to the future roadmap of this project?  ISO 15693 would support additional use cases.  I wasn't sure what the plans for this project were.

nehpetsde, 2022-02-25:
This is a pretty simple story as the readers don't support Type 5 Tag.

closed 2022-03-08 by nehpetsde