nfcpy/nfcpy #210: APDU splitting (Type4)

ikarus23, 2022-03-02:
Hi. First of all: Thanks for this great library!

I have an Android application and a Type4 tag. The Android app send the following APDU to the tag:
```
[84 C5 01 00 24] D6EEBB054D6F39BCBF4E14DBEE2145DDD219EFC9972FED9191DF027FD36A80B03658C5BD
```
When transmitted, Androids splits the APDU as shown in the this sniffed communication (including the ISO 14443 data):
```
Rdr |12  84  c5  01  00  24  d6  ee  bb  05  4d  6f  39  bc  bf  4e  14  db   |
    |ee  21  45  dd  d2  19  ef  c9  97  2f  ed  91  5a  87                   |
Tag |a2  e6  d7                                                               |
Rdr |03  91  df  02  7f  d3  6a  80  b0  36  58  c5  bd  47  f0               |
Tag |f2  09  d9  cc                                                           |
Rdr |f2  09  d9  cc                                                           |
Tag |03  90  00  2d  53                                                       |

```
As seen in the communication, the tag is fine with the APDU and responses with `9000`. However, when I try to send the same APDU with nfcpy, the APDU gets split differently (two bytes later): 
```
Rdr |13  84  c5  01  00  24  d6  ee  bb  05  4d  6f  39  bc  bf  4e  14  db   |
    |ee  21  45  dd  d2  19  ef  c9  97  2f  ed  91  91  df  ad  87           |
Rdr |b3  ee  d6                                                               |
Rdr |b3  ee  d6                                                               |
Rdr |b3  ee  d6                                                               |
```
As the communication trace shows the tag is not fine with this and does not respond anymore.

Is there a bug in nfcpy when it comes to splitting long APDUs? Or could it be the tag that does not work according to spec? I'm not sure how APDU splitting should work...

ikarus23, 2022-03-02:
I've just compared this to another "reader". I've used the Proxmark3 RFID tool to send the long APDU to the tag. The Proxmark does the splitting exactly like Android (and the tag responses with `9000`).

ikarus23, 2022-03-02:
Some more information: Because of the wrong(?) APDU splitting and the tag not responding to that, nfcpy shows the corresponding error messages:
```
2022-03-02 20:52:27,991 - nfc.tag.tt4 - WARNING: ISO-DEP timeout error (#1)
2022-03-02 20:52:28,425 - nfc.tag.tt4 - WARNING: ISO-DEP timeout error (#2)
2022-03-02 20:52:28,855 - nfc.tag.tt4 - WARNING: ISO-DEP timeout error (#3)
2022-03-02 20:52:29,286 - nfc.tag.tt4 - ERROR: ISO-DEP unrecoverable timeout error
```
Could it be that this is connected to https://github.com/nfcpy/nfcpy/issues/204 ?!?

nehpetsde, 2022-03-08:
Many thanks for reporting this issue. Indeed this is a bug in nfcpy's computation of the MIU for the ISO DEP protocol where additionally the two bytes checksum must subtracted for the correct information unit length. I was puzzled first why this could go unnoticed for so long but apparently all Tags I've tested effectively restrict the MIU to safe values via the NDEF management information.

closed 2022-03-08 by nehpetsde

ikarus23, 2022-03-08:
Hi @nehpetsde! Thank you for the quick response and the quick fix. I've just tried it and it worked like a charm!

Do you have any plans on when you will do a release (with this fix)?

nehpetsde, 2022-03-10:
@ikarus23 I've just released this fix with version 1.0.4. Many thanks again for your contribution.

PR #211 (2022-03-08): Use correct MIU for ISO DEP protocol.
The ISO DEP MIU must be 2 bytes shorter to account for the checksum added by the reader.

Fixes #210