nfcpy/nfcpy #214: New Sony NFC Reader/Writer  device RC-S300 is not supported

Cova8bitdots, 2022-03-18:
#TL;DR
Sony published new NFC reader/writer device `RC-S300` is currently not supported on nfcpy.
If we try to use it, nfcpy says `OSError: [Errno 19] Operation not supported by device` 

# What to do
- Add new value on device map such as `(0x054c, 0x0dc9): "rcs300",  # Sony RC-S300`
   https://github.com/nfcpy/nfcpy/blob/8a195262fe86167845d282ba5bc9cdaadf6441c4/src/nfc/clf/device.py#L42-L52 
- Add new script `rcs300.py` on https://github.com/nfcpy/nfcpy/tree/master/src/nfc/clf

Unfortunately, I'm not specialist about NFC devices, I have no idea how to research some device information of RC-S300
( ex, chipset, protocol , ...)
So, I want someone's help to support `RC-S300` to use.

Taposan0825, 2022-06-07:
Likewise, I too would like to use RC-S300 with nfcpy. Is it possible to analyze the device?

Taposan0825, 2022-06-07:
I would like to request an expert to investigate.