nfcpy/nfcpy #218: ACR122U Reader usb1.USBErrorNotSupported: LIBUSB_ERROR_NOT_SUPPORTED [-12] Error

SurpriseMF3000, 2022-08-04:
Hello,
i want to use nfcpy with the ACR122U Reader/ Writer but get the Error: `usb1.USBErrorNotSupported: LIBUSB_ERROR_NOT_SUPPORTED [-12]`

Code (NFC P2P): 
```
import nfc
import ndef
from threading import Thread

def beam(llc):
    snep_client = nfc.snep.SnepClient(llc)
    snep_client.put_records([ndef.UriRecord('http://nfcpy.org')])

def connected(llc):
    Thread(target=beam, args=(llc,)).start()
    return True

with nfc.ContactlessFrontend('usb') as clf:
    clf.connect(llcp={'on-connect': connected})
```

Error: `usb1.USBErrorNotSupported: LIBUSB_ERROR_NOT_SUPPORTED [-12]`

What do i need to change?
I already tried changing the Driver with Zadig to WinLib but then NFC Terminal shuts down and disconnects 1 second after code execution.

Setup:
Windows 10
ACR122U NFC Terminal

Thanks for help and answers!

nopeless, 2022-09-03:
I have the same issue
I went to asc.com.hk and downloaded the driver for this nfc card and also downloaded zadig and installed winusb

[image]

This is what I have in my device manager

and this is the error
```
usb1.USBErrorNotSupported: LIBUSB_ERROR_NOT_SUPPORTED [-12]
```

Kane-sigCare, 2023-06-05:
I am having the same issue and just tried downgrading to version 1.0.3 to see if it would work (#222). Same error message in both 1.0.3 and 1.0.4 unfortunately.

WyWlf, 2024-01-26:
stuck on this issue. Did anyone managed to find a fix for this?

Kane-sigCare, 2024-01-28:
> stuck on this issue. Did anyone managed to find a fix for this?

I never found a fix. I used [smartcard](https://pypi.org/project/smartcard/) instead.