nfcpy/nfcpy #222: timeout error from v1.0.4

robshep, 2022-09-14:
Hi,

Thanks for this great software.

I've just installed 1.0.4 and little utility has broken.

**Here's the stacktrace**

```
INFO:nfc.clf:searching for reader on path usb
INFO:nfc.clf:using ACS ACR122U PN532v1.6 at usb:001:012
Scanning...
WARNING:nfc.tag.tt4:ACS ACR122U on usb:001:012 does not support fsd 256
WARNING:nfc.tag.tt4:ACS ACR122U on usb:001:012 does not support fsc 256
WARNING:nfc.tag.tt4:ISO-DEP timeout error (#1)
WARNING:nfc.tag.tt4:ISO-DEP timeout error (#3)
WARNING:nfc.tag.tt4:ISO-DEP timeout error (#5)
ERROR:nfc.tag.tt4:ISO-DEP unrecoverable timeout error
Traceback (most recent call last):
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/clf/pn53x.py", line 667, in send_cmd_recv_rsp
    return self.chipset.in_communicate_thru(data, timeout+0.1)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/clf/pn53x.py", line 421, in in_communicate_thru
    self.chipset_error(data)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/clf/pn53x.py", line 124, in chipset_error
    raise Chipset.Error(errno, strerr)
nfc.clf.pn53x.Chipset.Error: Error 0x01: Time out, the Target has not answered

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/tag/tt4.py", line 95, in exchange
    data = self.clf.exchange(data, timeout)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/clf/__init__.py", line 1055, in exchange
    rcvd_data = exchange(self.target, send_data, timeout)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/clf/pn53x.py", line 671, in send_cmd_recv_rsp
    raise nfc.clf.TimeoutError
nfc.clf.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home2/rob/Dev/1/2/just_write.py", line 43, in <module>
    res = clf.connect(rdwr={'on-connect': on_tag, 'beep-on-connect':True}, terminate=after5s)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/clf/__init__.py", line 581, in connect
    result = self._rdwr_connect(rdwr_options, terminate)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/clf/__init__.py", line 612, in _rdwr_connect
    if options['on-connect'](tag):
  File "/home2/rob/Dev/1/2/just_write.py", line 29, in on_tag
    tag.ndef.records = [ndef_record]
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/tag/__init__.py", line 175, in records
    self.octets = b''.join(message_encoder(value))
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/tag/__init__.py", line 202, in octets
    self._write_ndef_data(data)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/tag/tt4.py", line 330, in _write_ndef_data
    offset += self._update_binary(offset, data[offset:])
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/tag/tt4.py", line 223, in _update_binary
    self.tag.send_apdu(0, 0xD6, p1, p2, data[:max_data])
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/tag/tt4.py", line 488, in send_apdu
    apdu = self.transceive(apdu)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/tag/tt4.py", line 437, in transceive
    data = self._dep.exchange(data, timeout)
  File "/home2/rob/Dev/1/2/.venv/lib/python3.10/site-packages/nfc/tag/tt4.py", line 116, in exchange
    raise Type4TagCommandError(nfc.tag.TIMEOUT_ERROR)
nfc.tag.tt4.Type4TagCommandError: unrecoverable timeout error
```
**Info after revert to 1.0.3**

```This is the 1.0.3 version of nfcpy run in Python 3.10.4
on Linux-5.15.0-47-generic-x86_64-with-glibc2.35
I'm now searching your system for contactless devices
** found ACS ACR122U PN532v1.6 at usb:001:012
```

**Application when reverted to 1.0.3**
```
INFO:nfc.clf:searching for reader on path usb
INFO:nfc.clf:using ACS ACR122U PN532v1.6 at usb:001:012
Scanning...
WARNING:nfc.tag.tt4:ACS ACR122U on usb:001:012 does not support fsd 256
WARNING:nfc.tag.tt4:ACS ACR122U on usb:001:012 does not support fsc 256
Written
```

I'd have a go at fixing it but I'm afraid I don't have any knowledge of low-level NFC
I thought I'd raise this here in case other people find the same 

Thanks again

Rob

nopeless, 2022-09-16:
@robshep do you think this is related to #218 ?

Kane-sigCare, 2023-06-05:
> @robshep do you think this is related to #218 ?

I am having the same issue as #218 and just tried downgrading to version 1.0.3 as a workaround.
Same error message in both 1.0.3 and 1.0.4 unfortunately.