nfcpy/nfcpy #226: Add support for `054c:1fc9 Sony Corp. NFC-PRO Reader` ?

ipaqmaster, 2023-01-04:
Doesn't seem to be part of the device.py array

Device info:

```
Bus 001 Device 014: ID 054c:1fc9 Sony Corp. NFC-PRO Reader
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x054c Sony Corp.
  idProduct          0x1fc9 
  bcdDevice            8.00
  iManufacturer           1 NXP INC.
  iProduct                2 NFC-PRO Reader
  iSerial                 3 56F1885E3132
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

ipaqmaster, 2023-01-04:
Naively tried shoehorning it in there to no avail:

```
$ python -m nfc -v -v -v
This is the 1.0.4 version of nfcpy run in Python 3.10.9
on Linux-6.1.1-arch1-1-x86_64-with-glibc2.36
I'm now searching your system for contactless devices
DEBUG:nfc.clf.transport:using libusb-1.0.26
DEBUG:nfc.clf.transport:path matches '^(usb|)$'
INFO:nfc.clf:searching for reader on path usb:001:016
DEBUG:nfc.clf.transport:using libusb-1.0.26
DEBUG:nfc.clf.transport:path matches '^usb(:[0-9]{1,3})(:[0-9]{1,3})?$'
DEBUG:nfc.clf.device:loading rcs380 driver for usb:054c:1fc9
ERROR:nfc.clf.transport:no bulk endpoints for read and write
DEBUG:nfc.clf.device:[Errno 19] No such device
I'm not trying serial devices because you haven't told me
-- add the option '--search-tty' to have me looking
-- but beware that this may break other serial devs
Sorry, but I couldn't find any contactless device
```

ipaqmaster, 2023-02-09:
I would not be surprised in the slightest if it's fake

mcirsta, 2023-02-10:
Mine works only with some software for Windows, NFC 122 I think it's called.
I opened it up as it only has 4 screws under the rubber feet.
Inside there's a FM17550 chip. This is a Chinese NFC chip. There's also a controller, MT006C6PB which is again some Chinese chip.
It's seen in Linux but I could not get it to work. Didn't try much though

albfan, 2023-04-18:
Is a chinese one: searching for same info in https://github.com/nfc-tools/libnfc/issues/693

@mcirsta thanks for info, I don't need to open case at least.

chip is documented here:

https://www.aliexpress.com/item/1005002458504365.html

```
1.1 Product introduction

FM17550 is a highly integrated non-contact communication chip that works at 13.56MHz and supports the following 4 different working modes.

Supports reader mode that complies with ISO/IEC 14443 TypeA protocol and Felica protocol

Support the reader mode conforming to ISO/IEC 14443 TypeB

Supports card simulation working mode that conforms to ISO/IEC 14443A protocol and Felica protocol

Support NFC working mode that complies with NFCIP-1 protocol

At the same time, it provides a low-power external card detection function, which is convenient for battery-powered, low-power operation, and requires real-time processing of external card reader devices that enter the radio frequency field at any time.

FM17550 has the characteristics of low voltage, low power consumption, strong drive capability, multi-interface support, and multi-protocol support. It is suitable for low-power, low-voltage, low-cost non-contact reader applications and NFC devices compatible with the NFC protocol.

1.2 Product features

Support ISO/IEC 14443 TypeA reader mode
Support ISO/IEC 14443 TypeB reader mode
Reader mode supports M1 encryption
Support ISO/IEC 14443, FeliCa card simulation working mode
ISO14443 TYPEA supports communication rate 106kbps, 212kbps, 424kbps, 848kbps
The communication rate based on FeliCa protocol supports 212kbps, 424kbps
The NFC communication rate that complies with the NFCIP-1 protocol is up to 424kbps
The operating distance of the reader can reach 50mm (depending on the antenna design)
Support NFC-WI interface
NFC-WI interface adopts independent power supply
Support multiple host interfaces
SPI interface up to 10Mbps
The I2C interface supports fast mode up to 400Kbps and high-speed mode up to 3.4Mbps
Serial UART interface, support RS232 frame format, maximum communication rate 1.2Mbps
Independent power supply for Host interface
64Byte transmit and receive buffer FIFO
The interrupt output mode is flexible and configurable
Built-in external field RF level detection
Automatic detection of internal integrated data mode (A/F automatic recognition during card simulation)
Multiple low power consumption modes
Soft powerdown mode
Hard powerdown mode
Deep powerdown mode (typical value 1uA)
Support low-power external card detection function
Programmable timer
Built-in oscillator circuit externally connected to 27.12MHz crystal
Wide voltage working range 2.2V~3.6V
The RF transmission driver is powered by an independent power supply, up to 5.5V
Built-in CRC coprocessor
Programmable I/O pins
```

But not sure how to use that info to read using nfcpy

thePanz, 2024-08-07:
@neomilium @albfan any updates on this device? how can we help?

albfan, 2024-08-07:
I guess if libnfc issue is closed, there's no match we can do here.

It's a clone, with no documentation available.

Using the provided nfc 122 app is the only way to use It.

Probably capturing USB communication with wireshark we might get some progress, but I don't know were to read about that, so I suggest to close this issue too

thePanz, 2024-08-07:
Thanks for the update :+1: 
sadly that would be the only way: sniffing packets and reverse-engineer the chip's protocol :disappointed: