nfcpy/nfcpy #231: NFC reader disconnect on completion

WarmWelcome, 2023-03-03:
Using reader acr122u-a9, the reader disconnects after any usage of the examples provided. To use them again, it forces me to replug them. I've attached an image showing the problem. My usage case for nfcpy requires that I have it constantly listening on the reader for tags, and needing to unplug and plug it back in constantly wouldn't do.
[image]

mcilhargey, 2023-06-02:
Likely the same issue as https://github.com/nfcpy/nfcpy/issues/154

Bluemanni, 2023-11-22:
I had the same issue on Ubuntu desktop.
The problem seems to the wakeup of the device after calling power_down in nfc/clf/pn532.py:232. (in the nfc module)
Removing this line fixed my problem.
However, without power_down, you better remove the reader if you don't need it for a longer period of time. :)