nfcpy/nfcpy #233: ModuleNotFoundError: No module named 'nfcpy'

Michdo93, 2023-05-29:
Hi,

I got following error when I try to import nfcpy:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'nfcpy'
```

I tried to install nfcpy in both Windows 10 and Ubuntu 20.04 under Python 2.7 and Python 3.8 respectively. I always get the same error. Although pip or pip3 has installed it completely.

I also went and cloned the repository and installed it accordingly from the directory.

I hope someone can help me. I think the Python 2.7 variant no longer works because libusb1 cannot be installed.

Kind regards

nehpetsde, 2023-05-30:
My guess is that you should "import nfc" and not nfcpy. Please see the first code sample on [https://nfcpy.readthedocs.io](https://nfcpy.readthedocs.io).

PanderMusubi, 2024-08-13:
@Michdo93 can this issue be closed?

exrector, 2025-03-09:
Same problem on macos
`exrector@MacBook-Pro-A NFC automation % python3 nfc_reader.py
Traceback (most recent call last):
  File "/Users/exrector/Downloads/NFC automation/nfc_reader.py", line 1, in <module>
    import nfc
ModuleNotFoundError: No module named 'nfc'`