nfcpy/nfcpy #236: NTag21x tags become irreversibly negatively modified when password protection is set

AlmirKadric, 2023-07-05:
For some reason nfcpy is setting the Capability Container bits which is irreversible as any writes here will result in an OR operation. This results in any password protected tags being irrevisbly locked out from being read by a passive devices such as an iPhone.

Though this could be a desired feature to prevent errors it should not be enforced and instead should be wrapped in an option argument.

https://github.com/nfcpy/nfcpy/blob/8a195262fe86167845d282ba5bc9cdaadf6441c4/src/nfc/tag/tt2_nxp.py#L167

Doctor-love, 2023-08-11:
Stumbled upon the same issue - have managed to "brick" a couple of tag this way :-/ . Would a workaround to prevent this behavior be to comment out line 431 to 435 or do you have another suggestion, @AlmirKadric  ?

Doctor-love, 2023-08-11:
Would also be happy to see a flag to prevent this behavior (or rather, prevent it by default and enable it with a flag), @nehpetsde

Doctor-love, 2023-08-14:
Anecdotally, it seems to work to comment out these lines (or rather 170-174, as I'm using Ultralight C tags) - don't get as nice error messages why I can't write to the tag from the NXP app, but at least it's possible to access NDEF records/trigger link opening on iOS/iPhones while utilizing the tag's authentication functionality.

I must however add that **I have no idea what I'm doing**, so if you ain't in a hurry better to wait until someone more knowledgeable comments on this issue.

nehpetsde, 2023-08-22:
Read and/or write protected NDEF compliant tags must properly indicate that state in the capability container. To deviate from the default behavior and create non-compliant NDEF tags, the protect function can be instructed to not touch the capability container bytes (page 3) with the argument `protect_from=4`.