nfcpy/nfcpy #240: Add support for Sony RC-S300/S1

csywssl, 2023-09-13:
According to the [Sony website ](https://www.sony.net/Products/felica/business/products/) the RC-S380/S will be discontinued. It appears the encouraged replacement device is RC-S300/S1. However, the RC-S380/S is based on the Port 400 chipset instead of the Port 100 chipset used in RC-S380/S. Therefore, it is likely a new driver file will need to be written for RC-S300/S1.

cminter, 2023-10-09:
I read a product page "Compatible with RC-S380/S: Can be used in applications developed for the RC-S380/S".
I just bought a Sony RC-S300 get started with `nfcpy`, but ran into this issue.

Since product info claims HW is compatible I tried hacking the library to see if I could add the device ID and have it just treat the device as previous.  `lsusb` shows as `ID 054c:0dc8 Sony Corp. FeliCa Port/PaSoRi 4.0`

Added device ID to `nfc/clf/device.py` with line `(0x054c, 0x0dc8): "rcs380",  # Sony RC-S300`.
Running `python -m nfc` gives error...

```
# python -m nfc
This is the 1.0.4 version of nfcpy run in Python 3.11.5
on Linux-6.4.15-100.fc37.x86_64-x86_64-with-glibc2.36
I'm now searching your system for contactless devices
ERROR:nfc.clf.rcs380:expected ack but got None
ERROR:nfc.clf.rcs380:expected ack but got None
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/site-packages/nfc/__main__.py", line 214, in <module>
    main(parser.parse_args())
  File "/usr/local/lib/python3.11/site-packages/nfc/__main__.py", line 60, in main
    clf = nfc.ContactlessFrontend(path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/nfc/clf/__init__.py", line 75, in __init__
    if path and not self.open(path):
                    ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/nfc/clf/__init__.py", line 149, in open
    self.device = device.connect(path)
                  ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/nfc/clf/device.py", line 90, in connect
    device = driver.init(transport.USB(bus, dev))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/nfc/clf/rcs380.py", line 982, in init
    chipset = Chipset(transport, logger=log)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/nfc/clf/rcs380.py", line 197, in __init__
    self.get_firmware_version()
  File "/usr/local/lib/python3.11/site-packages/nfc/clf/rcs380.py", line 335, in get_firmware_version
    log.debug("firmware version {1:x}.{0:02x}".format(*data))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: str.format() argument after * must be an iterable, not NoneType
```

Error is due to `get_firmware_version()` line `send_command(0x20, ...` as it returns `None`.  I commented the `log.debug()` out temporarily and tried returning `[0, 0]` but just then fails due to `get_pd_data_version` line `send_command(0x22, ...` for same reason
```
    def get_firmware_version(self, option=None):
        assert option in (None, 0x60, 0x61, 0x80)
        data = self.send_command(0x20, [option] if option else [])
        log.debug("firmware version {1:x}.{0:02x}".format(*data))
        return data

    def get_pd_data_version(self):
        data = self.send_command(0x22, [])
        log.debug("package data format {1:x}.{0:02x}".format(*data))
```

cminter, 2023-10-09:
Though I really don't know what I'm doing, I guess bigger issue is the earlier error logged from `nfc.clf.rcs380:send_command()`.
```
This is the 1.0.4 version of nfcpy run in Python 3.11.5
on Linux-6.4.15-100.fc37.x86_64-x86_64-with-glibc2.36
I'm now searching your system for contactless devices
ERROR:nfc.clf.rcs380:expected ack but got None
ERROR:nfc.clf.rcs380:expected ack but got None
```

cminter, 2023-10-09:
noticed this issue seems to be same as #214

csywssl, 2023-10-10:
@cminter I did the exact same steps as you and also hit the issue with `get_firmware_version()`. That's when I went back to the Sony website and found that RC-S300/S1 is based on the Port 400 chipset instead of the Port 100 chipset used in RC-S380/S. This makes me think that the drivers probably have significant changes under the hood, and Sony probably means that existing applications will still work if you are using the Sony SDK. 

FWIW, the RC-S300/S1 I purchased also shows device id `0x0dc8`